OctoPrint/OctoPrint #1576: [Request] Display latest successful print duration as metadata

Finkes, 2016-11-05:
I would like to see the duration of my last successful print of a certain object, so I know, how long it will take me to print it again. The estimation label could be used for that, as you will know more or less exactly how long it will take after you have printed it once.

The reason for this request is, that the estimation is often not that precise.

[image]

JohnOCFII, 2016-11-05:
@Finkes - The Print History plugin will give you the information you desire.  
http://plugins.octoprint.org/plugins/printhistory/

Finkes, 2016-11-05:
alright I will give it a try. thanks!

closed 2016-11-05 by Finkes