OctoPrint/OctoPrint #1578: Octoprint with pi3 communication timeouts.

Brianp82, 2016-11-06:
So I have a pi3 hooked up to my printer via a USB cable. I am powering the pi with a wall adapter that is 5v and 2.5A. It came with my pi3 kit. I have been having random communication errors. I have tried moving my comm timeout to 60 from 30 with no luck. It just waits 1 minute then says the comm timeout again 3 times and aborts. I need help!!! Please!! I apologize in advance if I do not have all the information needed right away. I am new to this.

#### What were you doing?

I am printing with Octoprint and getting commincation timeouts while printing and the printer just freezes at random times mid print. It wont start printing again. 

#### What did you expect to happen?
I expect my print to finish. lol? I have many successful prints but about 30% failed prints from comm error.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0rc1 (rc/devel branch) ---But also happened on the non dev firmware. Only upgraded to use color change plug.

#### Printer model & used firmware incl. version

Maker Select v2.1 - Not sure what firmware. Slicing with S3D

#### Browser and Version of Browser, Operating System running Browser

Latest version of google chrome. Windows 10 pro. This problem happens when browser is closed too.

#### Link to octoprint.log

https://gist.github.com/Brianp82/8c02057a27ab6a260958df2b56322264

#### Link to contents of terminal tab or serial.log

Serial log was empty. Couldnt get the contents of the terminal tab. Had to reboot fast to get control of my extruder away from my print.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

I dont know how to do this.

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

Brianp82, 2016-11-06:
If this helps I saved a prior log where it stopped printing. I am not sure if I just didnt wait until it said comm timeout(30 sec) and just shutdown the pi myself or I am pretty sure it shutdown by itself. Here is the log though for reference. 
https://gist.github.com/Brianp82/847f6d26f0ba8aad7a58b7bd7a034f8d

Brianp82, 2016-11-06:
I just got this too while preheating my extruder in idle state and getting ready to print. 

2016-11-06 01:15:03,886 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Session_Exo_2.gcode
2016-11-06 01:23:45,788 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Session_Exo_2.gcode
2016-11-06 01:26:35,975 - octoprint.util.comm - INFO - Timeout while in an active heatup, considering heatup to be over.
2016-11-06 01:28:36,017 - octoprint.util.comm - INFO - Communication timeout while idle, trying to trigger response from printer.
2016-11-06 01:30:36,119 - octoprint.util.comm - INFO - No response from printer after 3 consecutive communication timeouts, considering it dead.
2016-11-06 01:45:24,235 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.3
2016-11-06 01:51:45,416 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.3
2016-11-06 01:51:46,506 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.23s

ophilli, 2016-11-06:
Are you running OctoPi or did you install OctoPrint on your own?

Brianp82, 2016-11-06:
I installed Octopri image file and configured it to run with the pi.

Brianp82, 2016-11-06:
Maybe I am a little confused though. I know I installed OctoPi .13 from the website and followed the tutorial. It worked perfect. What do you mean did I install OctoPrint on my own? What is the difference

foosel, 2016-11-06:
From your log:

```
2016-11-05 17:19:19,815 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0rc1-py2.7.egg/octoprint/util/comm.py", line 1621, in _readline
    ret = self._serial.readline()
  File "build/bdist.linux-armv7l/egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

and also 

```
2016-11-05 17:28:17,224 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0rc1-py2.7.egg/octoprint/util/comm.py", line 1621, in _readline
    ret = self._serial.readline()
  File "build/bdist.linux-armv7l/egg/serial/serialposix.py", line 475, in read
    raise SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

and also 

```
2016-11-05 22:45:49,981 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2016-11-05 22:46:49,984 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2016-11-05 22:47:50,043 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2016-11-05 22:48:50,052 - octoprint.util.comm - INFO - No response from printer after 3 consecutive communication timeouts, considering it dead.
```

The communication between OctoPrint and your printer is not reliable, as you already pointed out too. Messages are getting lost and the two upper segments also hint at communication issues on the operating system level (so even outside OctoPrint's control). This is an issue between your Pi and your printer, and nothing OctoPrint can solve in any way. Check your cabling, try another (shorter) USB cable, ideally one with a ferrit bead. Also make sure you don't have it running next to motor or heater wires, or any other wires that might be carrying huge potentially changing currents. Finally - you have stated that you have a 2.5A wall adapter for your pi. That's all nice and well, but still try another one to rule that out as being the culprit here (I have heard from other users that ran into issues with reliable printer communication due to their Pi's PSUs). Writing "5V 2.5A" on a wall adapter is easy, that doesn't mean it actually is delivering both of that, and stable for that matter.

In #1574 you also provided this short segment from the terminal:

```
Send: N9 G92 E0*78
Recv: ok 9
Send: N10 G1 E-3.0000 F3000*41
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N11 M105*23
Recv: X:0.00 Y:0.00 Z:0.00 E:0.00
Recv: ok 10
```

That also indicates something in the firmware communication behaving wonky. It takes too long to reply to a simple `G1` extrude command, and when it replies it produces output I wouldn't expect as reply to a move (`X: ....` instead of only the `ok 10` that follows that). Considering the other two symptoms of printer communication issues you had in your log (quoted above), my guess is that it's not a case of a too short timeout, but of an actual unstable communication which - again - OctoPrint can do nothing about and you have to solve that first. So I'm not surprised that experimenting with the timeout length didn't help here. Your printer stops responding. It basically goes AWOL. OctoPrint can't keep up communication with a communication partner that refuses to reply.

About the reboot: Directly after that comm timeout message in `octoprint.log` you also have this:

```
2016-11-05 22:48:58,758 - octoprint.server.api.system - INFO - Performing command for custom:reboot: sudo shutdown -r now
2016-11-05 22:48:58,910 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.3
2016-11-05 22:48:58,952 - octoprint.server - INFO - Shutting down...
2016-11-05 22:48:58,987 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2016-11-05 22:48:58,988 - octoprint.events - INFO - Event loop shut down
2016-11-05 22:48:58,999 - octoprint.server - INFO - Goodbye!
```

That's a reboot triggered manually through the "System" menu. From the looks of it, you instructed OctoPrint to reboot the whole Pi directly after OctoPrint ran into the unrecoverable communication error (read: the firmware stopped replying). That would explain why it rebooted. As a general piece of information, no, you do not need to reboot the whole machine to make it reconnect to your printer, simply clicking on "Connect" in the Connection panel (upper left) will suffice after a disconnect caused by an error.

Now... after having all that analysed, based on what information you provided over the course of all those tickets and other outlets, your prints are getting interrupted by communication errors caused by unreliable communication between your Pi and your  [... cut ...] ng else. You need to solve this first before you can print via serial. I suggested a couple of things to try above (mostly to do with cabling). 

Your reboots are caused by you (or someone else) clicking "System > Reboot".

Neither is an issue within OctoPrint and hence does not actually belong into the bug tracker, which is why I'll mark this as closed. That doesn't mean that you won't be able to comment anymore (or other people for that matter), it's merely so that I don't lose track of tasks.

closed 2016-11-06 by foosel

ntoff, 2016-11-06:
What do you mean you can't connect to the server without rebooting? How are you rebooting the server (octoprint) if you can't connect to it? Do you mean octoprint can't connect to your printer? The browser won't load the octoprint page?

Brianp82, 2016-11-06:
So basically after it has a comm error I try to disconnect and reconnect and it wont actually connect. But I can go to reboot up top in octoprint and it will reboot the pi. Then it connects instantly. But sometimes I can actually disconnect and reconnect. I have tried 3 different usb cables. Even tried 250000 baud rate. I dont know what else to try. I just uploaded a file to SD from octo and printed it no problem from the printer. Now I am printing perfectly fine with the same USB cable from S3D directly wired to the printer. It ONLY happens with octoprint. Think the pi is bad?

andybali, 2018-04-02:
In about 50 print it happen 2 times by my printer. He stopped after 5 hours printing, 2 hours left. Next try with same g-code was OK.

 Recv: ok
Send: N177408 G0 X71.81 Y158.369*26
Recv: ok
Send: N177409 G0 X90.839 Y150.678*34
Recv: ok
Send: N177410 G0 X90.886 Y150.18*30
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Printing' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor

johny-mnemonic, 2019-01-17:
I can confirm, that what @foosel suggests is right:
>  Check your cabling, try another (shorter) USB cable, ideally one with a ferrit bead. Also make sure you don't have it running next to motor or heater wires, or any other wires that might be carrying huge potentially changing currents. 

My print just got broken by switching off a lamp which I use to have light on the bed.
The USB cable between printer and RPi is going directly over the A/C switch of that lamp, so the abrupt change of electric field when I switched the lamp probably was enough to brake the communication completely. At least my log looked exactly like what is in the post above.