OctoPrint/OctoPrint #1579: [Brainstorming] RFID & Database User Access Control

ophilli, 2016-11-06:
I administer several printers running Octoprint on a university network. We have an access control system for other tools in our space like lathes and band saws that uses RFID readers to detect a student ID card, and then poll a MySQL database for the students permissions. This system controls access by using a relay to turn the machine power off and on.

I would like to write a plugin / add-on for Octoprint that accepts the ID data from an RFID reader, checks the SQL database for permissions, and then signs the proper user into Octoprint. 

It doesn't look like there is an available plugin mixin or hook for access control at this time. From what I can find in the docs it looks like this might involve writing a new class to replace octoprint.users.FilebasedUserManager. Perhaps I can just write a plugin and simply change my config.yaml to point to something like octoprint.plugin.SQLbasedUserManager.

I'd greatly appreciate any guidance in figuring out how to best tackle this project!

I have read the FAQ.

Salandora, 2016-11-07:
Hi,

I'm also on this actually, nearly same situation. Students shall get limited access.
Actually the whole Permission system has not been developed further than admin and user permission.
This is the point where I start working.
Somewhere here is an already opened ticket about a granular permission system I'm using as a base.
After that it should be easy to implement an RFID reader plugin and give permissions to the users.

ophilli, 2017-01-09:
@Salandora sounds like #1110 might be the ticket you're referring to.

Currently any "User" can pause or cancel any other "Users" jobs. I wonder how easy it would be to specify that only the user that owns the job, or the admin could pause/cancel.

Salandora, 2017-01-09:
This would be easy to do but I think it is a security risk.
I would prefer to let everyone(except guests) cancel a print and log who was it in case of abuse.

I think of a scenario like this:
Person A starts a print and moves away.
Person B noticed the print went wrong and cancels it to save material and protect the printer.
If Person B would be unable to stop the printer (maybe because it is behind a closed door, or something) then you would waste material and maybe damage the printer.