OctoPrint/OctoPrint #1581: [Bug] Mac Chrome display unicode symbol as squares

mateddy, 2016-11-06:
#### What were you doing?

Open Octoprint webpage with Chrome with/without logging in

#### What did you expect to happen?

Unicode symbols displaying properly

#### What happened instead?

They first display properly,  

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0.dev1514+g4689a40 (devel branch)

#### Printer model & used firmware incl. version

Kossel Mini, but seems not relevant

#### Browser and Version of Browser, Operating System running Browser

Google Chrome Version 54.0.2840.71 (64-bit)
Mac OS X Sierra Version 10.12.1 (16B2555)

#### Link to octoprint.log

Not applicable

#### Link to contents of terminal tab or serial.log

Not applicable

#### Link to contents of Javascript console in the browser

http://pastebin.com/pxyyvXAG

#### Screenshot(s) showing the problem:

Attached.

I have read the FAQ.

[image]

mateddy, 2016-11-06:
Same issue not observed in Safari, Firefox or Windows Chrome; tried resetting Mac Chrome

foosel, 2016-11-07:
Those are not unicode symbols but a custom font, Font Awesome, which should be loaded along side when OctoPrint loads. If it's not loaded, it looks like that. Are there any 404 errors under the Network tab of the developer console (same location where you found the javascript console)?

I just tried this on the same setup on Browserstack, Chrome for Mac, version 54, under Sierra, and I could not reproduce it, so it doesn't appear to be a general compatibility problem between Font Awesome or OctoPrint and Chrome on Sierra

[image]

mateddy, 2016-11-07:
No 404.  In the bug report, I should have typed the following (may not be a good idea to edit it):

**What happened instead?**

They first display properly, (as Gina has shown) and then when the page loaded completely, most symbol was replaced by squares.

I tend to believe it's my chrome problem too.  Same problem is not reproduced in iOS Chrome, iOS Safari and Windows Edge.

JohnOCFII, 2016-11-07:
Just to add another datapoint, the font is displayed properly for me as well on Macintosh El Capitan as well as Sierra.  It also displays properly on iOS 10.  This is with Safari.

foosel, 2016-11-09:
@mateddy your javascript console is showing a lot of weird errors with regards to other fonts loaded from an external site: 

```
teddy.no-ip.com/:1 Failed to decode downloaded font: http://pstatic.bestpriceninja.com/nwp/v0_0_1138/release/Extras/Fonticons/foicons.woff?74188094
teddy.no-ip.com/:1 OTS parsing error: incorrect file size in WOFF header
teddy.no-ip.com/:1 Failed to decode downloaded font: http://pstatic.bestpriceninja.com/nwp/v0_0_1138/release/Extras/Fonticons/foicons.ttf?74188094
teddy.no-ip.com/:1 OTS parsing error: incorrect entrySelector for table directory
teddy.no-ip.com/:1 Failed to decode downloaded font: http://pstatic.bestpriceninja.com/nwp/v0_0_1138/release/Extras/Fonticons/foicons.woff?74188094
teddy.no-ip.com/:1 OTS parsing error: incorrect file size in WOFF header
teddy.no-ip.com/:1 Failed to decode downloaded font: http://pstatic.bestpriceninja.com/nwp/v0_0_1138/release/Extras/Fonticons/foicons.ttf?74188094
teddy.no-ip.com/:1 OTS parsing error: incorrect entrySelector for table directory
teddy.no-ip.com/:1 Failed to decode downloaded font: http://pstatic.bestpriceninja.com/nwp/v0_0_1138/release/Extras/Fonticons/foicons.woff?74188094
teddy.no-ip.com/:1 OTS parsing error: incorrect file size in WOFF header
teddy.no-ip.com/:1 Failed to decode downloaded font: http://pstatic.bestpriceninja.com/nwp/v0_0_1138/release/Extras/Fonticons/foicons.ttf?74188094
```

Some rogue extension? Have you tried in incognito mode/with a clean profile?

mateddy, 2016-11-09:
I tried cleaning my Chrome before submitting the report; I just tried incognito mode and the problem still persists.  Maybe there is some problem with my computer font.

Let me try on another Mac/Chrome next week.

mateddy, 2016-11-14:
Issue not reproduced on another mac/chrome, so I guess I should close this issue.

closed 2016-11-14 by mateddy