OctoPrint/OctoPrint #1582: A nice feature may be

RJ81, 2016-11-06:
A nice feature may be to have a thumbnail image of the parts stored in the memory.  I mocked-up a partial screenshot to show what I mean.

[image]

VanKurt, 2017-11-16:
+1. This would be really awesome.

BTW: RepetierHost already has this feature implemented.

dzenbot, 2018-01-03:
Sounds like a great plug-in candidate!

RJ81, 2018-01-03:
That would be nice. 

Still would appreciate any instructions to correct this error

Richard

This message was sent from my iPhone. Please excuse any briefness or misspelled words in the reply.

eneski, 2019-01-26:
I am also looking for something like this.. someone should definetely do it

fieldOfView, 2020-01-06:
The Ultimaker Format Package plugin does this, in combination with Ultimaker Cura:
https://plugins.octoprint.org/plugins/UltimakerFormatPackage/

LazeMSS, 2021-03-31:
Plugins are available for this:
https://plugins.octoprint.org/plugins/prusaslicerthumbnails/
https://plugins.octoprint.org/plugins/UltimakerFormatPackage/

closed 2021-03-31 by LazeMSS