OctoPrint/OctoPrint #1585: Print Started Before Hotend Reached Temp

jpvonhemel, 2016-11-08:
#### What were you doing?

Printing second copy of the following GCode.   

https://gist.github.com/jpvonhemel/b509f9fdc8e10dfd0bd1d59d204c45dc

Printer was idle/cool at the end of the prior print of the same model, replaced bed glass and pressed the Print button to print the same design as prior print.

#### What did you expect to happen?

I expected the bed to heat, which it did heat to 70C, followed by the hotend, and finally start the print.

#### What happened instead?

After the bed reached temperature of 70C, the print started before the hot end reached temp.   Startup script was ran.  Extrusions were cold and I had to stop the print. 

#### Branch & Commit or Version of OctoPrint

Version: 1.2.17 (master branch)

#### Printer model & used firmware incl. version

Makergear M2 Dual Firmware

#### Browser and Version of Browser, Operating System running Browser

Chrome 54.0.2840.87

#### Link to octoprint.log

https://gist.github.com/jpvonhemel/783bb1fe3fbc6646a46ac6cf7738df9d

#### Link to contents of terminal tab or serial.log

https://gist.github.com/jpvonhemel/fd815dfb7a35905aa855ef94b348c3d8

#### Link to contents of Javascript console in the browser

https://gist.github.com/jpvonhemel/9087edd6fbc1407392008a03ff615b5c

I have read the FAQ.

jpvonhemel, 2016-11-08:
I just pressed Print and again, it is starting the print before the hotend starts to heat up.

jpvonhemel, 2016-11-09:
Looks like the problem is not a bug in Octoprint, but a setting in Simplify3D that became unchecked somehow.    Sorry to have opened this by mistake.   Love Love Love Octoprint!

[image]

jpvonhemel, 2016-11-09:
Would love to know where in the gcode this slicer setting is placed.

FHeilmann, 2016-11-09:
There are two sets of G-Codes for heatup.

`M140` and `M190` for the heated bed and
`M104` and `M109` for the hotend.

`M104` and `M140` set the temperature but return control to the host immediately. This means that G-Code processing continues immediately and if your hotend/bed are not at temperature at this point you'll run into problems.

`M109` and `M190` set the temperature and block until said temperature is reached to a configured precision (configurable with the Marlin config, default values are fine though). This ensures that the G-Code processing will not continue until everything is up to temperature.

In your G-Code file in lines [188-190](https://gist.github.com/jpvonhemel/b509f9fdc8e10dfd0bd1d59d204c45dc#file-flower-gcode-L188-L190) you can see that both `M140` and `M190` are used for the bed (the `M140` is kinda useless here because the following command will wait for the temperature to be reached anyway) but only a `M104` is issued for the hotend, so the printer won't wait for the hotend temperature to be reached. Checking the setting in Simplify3D should replace the `M104` with a `M109` which solves your problem. Alternatively, you can put it in your `start.gcode` file which will cause Simplify3D to suppress its own temperature G-Code placement. Read [this](https://forum.simplify3d.com/viewtopic.php?f=8&t=1959) forum post on instructions how to place heater commands in the start.gcode that automatically places the correct temperature from the Temperature settings tab.

So why have two sets of commands?

Blocking heating using `M109` and `M190` is desired before starting the print to ensure the heaters have reached their desired temperature before doing anything. However the blocking behavior is problematic during a print if you want to adjust your temperature up/down during the print and the printer sits there for however long ti takes to reach the new temperature and oozes/burns up your existing layers. That's what the two non-blocking commands `M104` and `M140`  are for.

Some people also use the non-blocking variants before the blocking variants to allow both heaters to heat up in parallel. When only using `M109` and `M190` you would have to wait for the hotend to heat up AND THEN the bed to heatup, whereas when issuing  `M104` and `M140` followed by `M109` and `M190` both heaters will heat in parallel which will reduce the total time required to heat up. People who don't have issues with deformation/expansion of their hardware also issue `M104` and `M140`, perform the bed leveling procedures (G28, G29) and then issue `M109` and `M190` which saves some time.

This post has gotten out of hand, but I hope it clarifies everything for you.

Cheers!

foosel, 2016-11-09:
@FHeilmann thanks a lot for the explanation, that just saved me some time! :)

@jpvonhemel as general rule of thumb, what OctoPrint does when you print a file is open the file and stream anything therein to your printer. So everything for your print job to be successful should be in your file. OctoPrint will never take control over your printer's heaters or the print head unless you tell it to yourself (by using the manual controls OR through a streamed GCODE file). That's also in the FAQ. So if temperature or head are behaving weirdly, always look at your slicer's settings/your GCODE files and your firmware (if the firmware isn't properly executed the commands it gets sent, there's also nothing OctoPrint can do about that).

For reference, the issue in your provided `serial.log` can be found [on this line](https://gist.github.com/jpvonhemel/fd815dfb7a35905aa855ef94b348c3d8#file-serial-log-L299). That's an `M104` instead of an `M109`, hence the firmware immediately returns control and doesn't wait for the hotend to reach temperature, as explained by @FHeilmann above.

Closing this since it's not anything OctoPrint can do anything about.

closed 2016-11-09 by foosel

jpvonhemel, 2016-11-09:
Thank you both for explaining this to me.  Sorry to have opened the ticket by mistake.

sorren4, 2018-03-14:
I know that foosel closed this issue and the last post explains what OctoPrint is "supposed" to do.  However, I've checked my Gcode and it has both the 104/140 and the 109/190 codes in it.  I've selected the "Wait for temperature controller to stabilize before beginning build" in S3D so it puts those lines in the Gcode.  However, when I upload the files to OctoPrint and tell the printer to start, it doesn't wait for the bed and nozzle to heat up (per the Gcode).  So something must still be off, or was maybe overlooked/changed in more recent versions.  Some of the competitive products I've tried don't have this issue but I prefer OctoPrint for its capabilities.

FHeilmann, 2018-03-15:
can you provide the gcode file that causes these issues?
Also please state your octoprint version and which firmware you are using

brettkp, 2019-01-27:
I'm seeing the same behaviour in OctoPrint 1.3.10. I've uploaded an example gcode file and the terminal commands from Octoprint to show what it's doing. I've had to increase the timeouts significantly otherwise it loses comes during the bed/extruder warm up and causes the print to fail.

My printer is the Kodama Trinus running Marlin fw.

[Octoprint_log.txt](https://github.com/foosel/OctoPrint/files/2799908/Octoprint_log.txt)

[Table cover clip v1_0.2_060 - Copy.gcode.txt](https://github.com/foosel/OctoPrint/files/2799907/Table.cover.clip.v1_0.2_060.-.Copy.gcode.txt)

Thanks,
Brett

foosel, 2019-01-28:
@brettkp from the looks of your terminal tab output there's nothing that OctoPrint does here that would be wrong. It sends your GCODE to the printer just as contained in the GCODE file. It's your printer's firmware's responsibility to actually perform a *blocking* heatup on am `M190` and an `M109`, and the output doesn't suggest an actual heatup sequence - since this isn't a `serial.log` with timestamps though it's hard to say if there is a sufficient pause between sending the `M109` and the following `ok`. 

I suggest you verify that in case of an issued `M109 S100` (for example) your firmware actually does block until the hotend reaches 100°C and only returns control back to the connected host with a final `ok` once that is the case and not before. Same with the bed and an `M190 S60` (for example). If it returns control before, the problem is your firmware not behaving as it should and you need to address that either with your printer's vendor or flash some firmware that *does* behave correctly.