OctoPrint/OctoPrint #1587: Cancel not taking effect until Bed/Extruder are fully heated

Jademalo, 2016-11-11:
#### What were you doing?
Pressing the "Cancel" button

#### What did you expect to happen?
The print to stop, and the bed and extruder to be disabled and cool down.

#### What happened instead?
The print continued until the bed was up to temperature, then it ran the afterPrintCancelled script.

#### Branch & Commit or Version of OctoPrint
1.2.17 (Master Branch)

#### Link to octoprint.log
http://pastebin.com/EvsQxe4J

I have read the FAQ.

I started a print, but realised there was an issue almost immediately and clicked cancel to stop the bed heating up.
When I pressed "Cancel" to stop the print, the bed continued to heat up. It only ran the afterPrintCancelled script when the bed was correctly up to temperature and "ok" was recieved. This essentially meant that the printer was on and needlessly heating up for approximately 8 minutes, which was all extremely unnecessary.

foosel, 2016-11-11:
Yes, and this can't be changed for let's say 99% of printers out there. The blocking heat commands M109, M190 and M116 are all truly blocking in the sense that the firmware will refuse any commands sent to it while they are active. There is a new feature in the current Marlin RC AFAIK which allows a specific command to be sent to stop heatup, but that is not wide spread enough that it can be depended on. 

For all other firmware variants, once you start a blocking heat the printer's firmware will simply stop accepting commands until that is done (apart from the emergency stop maybe, and that would be the wrong one to use here since it would completely bollock the printer for God until a manual per cycle). So there's no way for OctoPrint to tell it to please stop heating up now in case of a cancel.

It's sadly not the only case where firmware design didn't rally take connected hosts into account.

foosel, 2016-11-11:
Closing as duplicate of #1504

closed 2016-11-11 by foosel