OctoPrint/OctoPrint #1588: Feedback controls are broken in 1.3.0

foosel, 2016-11-11:
### Problem

Feedback controls such as

```
- command: M114
  name: Get Position
  regex: X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)
  template: 'Position: X={0}, Y={1}, Z={2}, E={3}'
```

no longer produce the request output. On closer look, they are not properly enriched with the expected additional regex processing data on retrieval from the settings backend and hence ignored by the actual feedback matching code.

Reason is that the default preprocessors are not properly applied in the current settings implementation. This error was introduced when refactoring those parts for settings overlay support.

### Solution

Make sure the default preprocessors are utilized on get and set operations if no specific ones are provided.

### See also

  * [this comment + follow up discussion](https://github.com/foosel/OctoPrint/issues/882#issuecomment-259444243) in #882

foosel, 2016-11-25:
1.3.0rc2 is out

MoonshineSG, 2016-11-26:
hmmm.... tried this on 1.3.0 RC2 and doesn't seams to work...

foosel, 2016-11-28:
@ovidiu I saw you commented that it didn't work, but the comment vanished - did it work out after all then and you deleted it or what happened there? :) It definitely does work for me...

MoonshineSG, 2016-11-28:
Yes, it appears I can type faster than my printer can respond. Sorry about that. It works.

foosel, 2016-12-01:
Closing this then, as it's fixed with 1.3.0rc2 (and I actually had intended to close it with my corresponding comment, don't know what went wrong there)

commit 5446dc64 (2016-11-11): Use default settings preprocessors if none are provided

Re-enables feedback controls.

Closes #1588
closed 2016-12-01 by foosel