OctoPrint/OctoPrint #1589: [Request] Update Tempurature When Pre-Heating Bed Before Print

sorvis, 2016-11-11:
I've noticing that when I use the Temperature tab to preheat the bed or extruder before I do a print the temperature graph does not work. 

It seems like during a print the M105 G-Code is sent. Just now I was playing with it and I found that if during the prehead of my bed if I manually send M105 then the graph updates very nicely. 

What would the effort be to call M105 every so often after a user has manually set the temperature before a print starts? 

Another idea I had was maybe a plugin could be made that puts a refresh button in the bar at the top where you could just click it if you cared about what the temp was.

Thanks for all the great work on OctoPrint it is really great software!

foosel, 2016-11-14:
`M105` will always be sent, even outside of printing, unless a command known to be blocking the whole print communication is currently active. Among these commands are the heat-up-and-wait commands `M109` and `M190`. Those *should* be reporting back the current temperature every second though until the target is reached, but that depends on the firmware.

The behaviour you are describing sounds like something encountered with printers running Repetier Firmware where not all features marked "Repetier" are checked in OctoPrint under Settings > Features.

To verify this is the case here (and this is not in fact a bug), could you please provide a [`serial.log`](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#where-can-i-find-those-log-files-you-keep-talking-about) of your printer communication?

sorvis, 2016-11-17:
Here is the link to my [serial.log](https://gist.github.com/sorvis/c1875308a2f19579091a8f1d26bd15dd) file.

So I tried enabling all of the settings marked with Repetier in the settings you mentioned and then rebooted the server then suddenly the temperature seems to be working. I'm using an i3 from Maker Select.

It sounds like then I've hit an already known bug?

Thank you very much for taking the time to look at my issue.

foosel, 2016-11-17:
> It sounds like then I've hit an already known bug?

No, not a bug, an incompatibility between firmwares that requires you to tell OctoPrint to do a couple of things slightly differently. 

I could probably rant hours about the state of protocol compatibility (or the lack thereof for that matter) across the printer firmware landscape. The part that's important for you is simply: if it's Repetier firmware, check everything that is marked as Repetier, or stuff will not work as expected.

> Here is the link to my serial.log file.

That's `octoprint.log` ;) But considering that your problem appears to have been exactly what I suspected since stuff started working once you checked the Repetier options, I guess we won't need `serial.log` anymore.

I'll be marking this as closed now, thanks for reporting back!

closed 2016-11-17 by foosel