OctoPrint/OctoPrint #1591: Filename of timelapse files is missing leading zeros

sanchosk, 2016-11-13:
#### What were you doing?

I was printing regular file as always.

#### What did you expect to happen?

I expected the timelapse interim files to be named from 00001 to xxxxx.

#### What happened instead?

The files name does not have the leading zeros. While the final timelapse video is produced OK, I cannot watch the interim process. Normally I was able to issue a command like "mplayer mf://*.jpg -loop 0" in tmp folder, now it only plays few frames due to the numbering. While I can run a simple bash script to rename the files, I am afraid this would negatively impact the final timelapse generation.

#### Branch & Commit or Version of OctoPrint

1.3.0rc1

#### Printer model & used firmware incl. version

RepRap i3, Chinese clone.

#### Browser and Version of Browser, Operating System running Browser

Irrelevant.

#### Link to octoprint.log

Irrelevant for this issue.

#### Link to contents of terminal tab or serial.log

Irrelevant for this issue.

#### Link to contents of Javascript console in the browser

Irrelevant for this issue.

#### Screenshot(s) showing the problem:

N/A

Yes, I have read the FAQ.

foosel, 2016-11-14:
I got rid of the leading zeros since that would cause issues if the number of timelapse frames would actually exceed 99999 stills.

However, I see where you are coming from and hence just spent a bit of time experimenting. The behaviour that I'm seeing is not that it only plays a few frames, but that it plays all of the frames, however they are out of order since the `*` matches alphabetically, making "image-10.jpg" come before "image-2.jpg".

I took a look into mplayer's source and using `%d` instead of `*` fixes that behaviour. It necessitates the name of the timelapse session to work though (which you should include anyhow, because otherwise you'll stitch together all unrendered timelapses) since you can't have two wildcards in place:

    mplayer mf://some-timelapse_timestamp_%d.jpg -mf fps=25:type=jpg

or, as a more concrete example (and just tested with an actual set of unrendered timelapse stills of a print currently going on):

    mplayer mf://lenovo-clamp-dell_20161114083149-%d.jpg -mf fps=25:type=jpg

Works fine over here and gives me a preview of the timelapse so far. Dumping this into a file `display.sh`:

``` bash
#!/bin/bash

PREFIX = $1
mplayer mf://$1-%d.jpg -mf fps=25:type=jpg
```

and `chmod +x` that allowed me to just enter `./display.sh leno<TAB><BACKSPACE><ENTER>` and - thanks to tab completion - have it render the correct dump of timelapse stills from the directory, in the correct order.

Does this solve your problem too?

sanchosk, 2016-11-14:
Ha, it never occurred to me that mplayer supports %d!
Thanks.
I'll test it in the evening, but as you already did, I am closing this as no issue.

closed 2016-11-14 by sanchosk