OctoPrint/OctoPrint #1592: [Brainstorming] Fan speed control

ntoff, 2016-11-15:
Sparked by a discussion in google+, it'd be nice to have the parts cooling fan speed adjustable rather than just an on / off button.

[image]

I've managed to make it work but as I have no clue what I'm doing with code, I'm 99% sure I'm not doing it the right / best / most efficient way. Is this something worthy of being a standard feature? Personally I'd use it all the time since my maximum fan speed is only ever about 30% which can easily be set in the slicer before printing but when manual control is required (say I forget to enable the fan in the slicer), I have to figure out what 30% is for the PWM value and enter it in the terminal.

foosel, 2016-11-15:
> Is this something worthy of being a standard feature?

Yes. But I'm thinking if it would make sense to allow the user to decide whether they want the old binary approach or a slider + off button via a feature flag. The reason being that while there's probably a large group like your's who would utilize it a lot because, there's probably an equally if not larger group which never used in-between values for the fan (I count myself to those actually, because it simply doesn't make a difference for my own prints).

>  I'm 99% sure I'm not doing it the right / best / most efficient way

I saw the [discussion on G+](https://plus.google.com/u/0/100879221424009989942/posts/bS8LC7iG4Mw) and the code snippets there. If what you have is basically what was posted there, yeah, I'd probably suggest moving the calculation logic from the template file into the view model (JS) file, as a `ko.computed` value that could then be utilized in the label of the button and for the actual command, similar to how the sliders for the feed and flow rate adjustment are done.

ntoff, 2016-11-16:
I did move it out of the template file with the intention of just copying the flow and feed rate thing but I didn't use a ko.computed and... well... I still don't like it. I just cheated and used the custom command sender.

https://github.com/ntoff/OctoPrint/commit/e8544a6dc117bbf99ebd3a0326d668284c161f36

>Yes. But I'm thinking if it would make sense to allow the user to decide whether they want the old binary approach or a slider + off button via a feature flag. 

I don't know how to make it a setting, I tried once before to add a setting for the motor off command since M18 does nothing for Repetier firmware but I didn't get very far. My knowledge of coding is pretty limited. I guess close this for now unless the ghost of coders past come to visit me in my sleep and I somehow gain all the knowledge of the coding masters, I doubt there's much more I can add to it. :(

foosel, 2016-11-16:
Don't close it, if push comes to shove we'll just turn it into a request ticket :)

Are you interested in getting a bit more coaching in how to implement this and giving it a try yourself? I'd be happy to explain a bit in depth how and why to do the things I suggested above!

ntoff, 2016-11-18:
I looked at the documentation to ko.compute and tried to have a go at it but I think it's all way above my  head. Thanks for the offer of coaching but I don't think I'll ever fully understand *how* to do it, even though I may understand *why*. I do have moments of inspiration when somehow things I make "just work" but those moments are few and far between with me being distracted by something shiny the more  common outcome. I do appreciate the offer but just think I'll never really understand it, I guess my brain just doesn't work that way.

ntoff, 2016-11-18:
Ok after wrestling with git (I know nothing) I cleaned up the multiple commits and have just one showing all the changes https://github.com/ntoff/OctoPrint/commit/51409a60c264f51e1b152132d622ef1b7e0b62ad

Am I at least somewhere close? The slider is a "Features" option (and it's off by default so standard on/off buttons are the default setting).

foosel, 2016-11-19:
Yes. In fact on first look I only see one thing I'd change and one thing that could be improved a bit (and that is minor stuff really!). I'm on mobile right now though, so typing that up will have to wait, possibly until Monday, sorry. Still wanted to give you a big :+1: though, you did it! :)

foosel, 2016-11-21:
I added my one comment to your commit (that was the "one thing I'd change" one, I can't for the life of me remember what I would have improved right now, looks fully fine to me as is), see [here](https://github.com/ntoff/OctoPrint/commit/51409a60c264f51e1b152132d622ef1b7e0b62ad#commitcomment-19903675).

ntoff, 2016-11-22:
Ok, changed the name of the variables https://github.com/ntoff/OctoPrint/commit/97dde221df28e949a070300cabb9ff9cbb52f460

One thing that I'll hate myself for asking but should the command be sent from elsewhere? As it is, it's being sent from control.js as an M code with the variable tacked on. Is there a better way to do that? Perhaps another file somewhere that just handles M and G codes? Looking at the way the other functions in there work (for example the flow rate), would making the fan speed slider work the same way involve writing another function into the API?

foosel, 2016-11-22:
> should the command be sent from elsewhere?

It's a bit ugly, but it's always been a bit ugly in that regard. I migrated the motion commands (so jog, extrude, home, flow and feed rate modifiers) a while ago, but back then I didn't bother with the fan and power control because I was (and still am) undecided if I shouldn't actually move that out into a pluggable component, and so it's as it is.

I don't think we should tackle that now on top of the slider feature, which basically acts as a drop-in replacement for the old version (which also codes the M code into the click definition).

If you want though (and only if you want), what you could do is turn the `setFanSpeed` method you introduced from the current implementation to something like this:

``` javascript
self.sendFanSpeedCommand = function (speed) {
    if (speed === undefined) {
        speed = self.fanPWM();
    }
    OctoPrint.control.sendGcode("M106 S" + speed);
};
```

Basically allowing to use the same method for setting specific manual values, or the current slider selection if now value is provided. And then adjusting the template to utilize that for the hardcoded "Fan on" and "Fan off" levels by calling it with 255 or 0. That way the `M106` command would only live at one point in the frontend code instead of being spread across three locations (and come to think of it, that might actually be the "improvement suggestion" I forgot between the weekend and yesterday).

ntoff, 2016-11-23:
On a whim last night I tried out the M33 Fio plugin which, among other things, adds its own buttons to the controls area. This slider addition kind of half breaks the buttons the M33 plugin adds if the slider control is enabled, a lone slider sits amongst the buttons and appears to do nothing (although M33 Fio's buttons seem to still work). I guess it needs more testing. Maybe the fan slider should just remain a custom user added button or a plugin and leave the fan controls the way they are.

closed 2016-11-25 by ntoff