OctoPrint/OctoPrint #1593: UI > System Menu | Items in Menu Duplicated

theunruly, 2016-11-18:
Latest dev Octoprint showing duped menu for system. Uninstalled TouchUI Plugin as test, items still duped. 
#### What were you doing?
Header Menu in UI > System Menu > Items are duplicated
Restart OctoPrint
Reboot
Shutdown
-------------------
Shutdown
Reboot
Restart Octoprint

#### What did you expect to happen?
Items to not be duplicated

#### What happened instead?
Duplicated Items

#### Branch & Commit or Version of OctoPrint
Version 1.3.0rc1 (rc/devel branch)

#### Printer model & used firmware incl. version
Tevo Tarantula - Marlin 1.1 rc7
MP Select Mini - Maylan 22.39

#### Browser and Version of Browser, Operating System running Browser
Chrome Version 54.0.2840.99 m Windows 10
Reproducible on 4 separate computers 2 networks.

#### Link to octoprint.log
http://pastebin.com/Viy0KZUE

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

[image]

I have read the FAQ.

AlexanderPro29, 2016-11-18:
I also want to note that these items are not translated when changing languages.

foosel, 2016-11-18:
To quote the [release announcement on the OctoBlog](http://octoprint.org/blog/2016/10/19/first-preview-of-1.3.0/):

> If you already had the system commands configured manually or had gotten them preconfigured through OctoPi, they’ll now show up twice - this can be rectified by simply removing the manually configured system commands, but you might want to wait with that until the stable release of 1.3.0 :) I might also still add some migration step for that. 

So that is a known issue.

The entries (from core, not from `config.yaml`) should adjust depending on selected language however, and that appears to be broken.

foosel, 2016-11-18:
Translation issue has been fixed on `devel` via the above three commits.

Rephrasing ticket to make it a request for (attempting) system command migration on start, at least for the redundant versions rolled out with OctoPi/suggested in the official setup guide. No way to migrate arbitrary (renamed & changed) system commands for restarting/rebooting.

foosel, 2016-11-18:
That's now implemented on `devel` and will be in 1.3.0rc2. See commits fc06c91ae86b1e9c1a2a27950ee6c2c0c06e760c and 71581c9c93528a4bab933c947b200c9fe428e3c1

theunruly, 2016-11-18:
I missed that blog entry, apologize for the duplicate post (see what I did there? lol) ... thanks!

foosel, 2016-11-25:
1.3.0rc2 is out

closed 2016-11-25 by foosel