OctoPrint/OctoPrint #1594: [Request]

ss89, 2016-11-20:
Hello,

i would like to request the ability of using variables on input fields in the configuration overlay.
My current need only for the webcam page, as my printservers ip adress changes a lot (webcam server is the same machine).

If that is for some reason (security?) not possible, would it be possible to make octoprint stream the webcam picture to the client instead of letting the webcam server contained in the configuration do it? That way i could just enter 127.0.0.1/localhost as ip which would then always work.

Have a great day/weekend!

ntoff, 2016-11-20:
If you're running it on a raspberry pi you could set up haproxy https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#make-everything-accessible-on-port-80 and the webcam url becomes a relative one

foosel, 2016-11-21:
What @ntoff said. Variables in the config won't solve your problem (your server might be multi homed, accessible under multiple names - which one to use?), using a reverse proxy that makes OctoPrint and webcam server available under the same host and port and hence allow for relative URLs however will, which is why that is exactly how OctoPi ships.

PS: that request is missing a subject - please add one next time :)

closed 2016-11-21 by foosel

ss89, 2016-11-21:
@ntoff @foosel thanks a lot for the idea. Will use nginx for that (should it let me)

foosel, 2016-11-21:
https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples#nginx :)