OctoPrint/OctoPrint #1597: Installation on OpenWrt fails.

lonelymyp, 2016-11-22:
During installation, an error occurs, it is impossible to install the necessary dependencies. in older versions of the installation took place without problems.
In version 1.1 there was no error.

```
Processing dependencies for OctoPrint==1.2.17 
Searching for psutil>=3.2.1,<3.3 
Reading https://pypi.python.org/simple/psutil/ 
Best match: psutil 3.2.2 
Downloading https://pypi.python.org/packages/dc/b2/ ... c3f8f67976 
Processing psutil-3.2.2.tar.gz 
Writing /tmp/easy_install-mNXmJS/psutil-3.2.2/setup.cfg 
Running psutil-3.2.2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-mNXmJS/psutil-3.2.2/egg-dist-tmp-MlvqMv 
warning: no previously-included files matching '*' found under directory 'docs/_build' 
warning: manifest_maker: MANIFEST.in, line 18: 'recursive-include' expects <dir> <pattern1> <pattern2> ... 

warning: build_py: byte-compiling is disabled, skipping. 

unable to execute 'ccache_cc': No such file or directory 
error: Setup script exited with error: command 'ccache_cc' failed with exit status 1
```
Any ideas what could be the problem?

markwal, 2016-11-30:
pip says that the problem with OpenWrt is that it can't install a suitable version of psutil, a python module required by, but not provided by OctoPrint.  It looks like there is something wrong with your build essentials on that router since I suspect the message about ccache_cc is indicating that it can't find the compiler indicated by the environment settings.

While sorting that out, you could could try "pip install -v psutil=3.2.2" as a quicker way to reproduce the problem.  Perhaps in your previous version all the required modules were available as wheels or already installed on your router?

lonelymyp, 2016-12-01:
In the old version of the OctoPrint module is not needed at all. it is not in the dependencies.
Command "pip install -v psutil=3.2.2"  gives the same error "unable to execute 'ccache_cc'"

OpenWrt 15.05 clean installation, additional modules are installed for the USB flash drive.
You know what compiler is needed?

foosel, 2016-12-01:
Probably a C compiler.

psutil is now needed for some disk space monitoring.

foosel, 2017-03-08:
Since I didn't hear anything back, considering this solved itself in a python environment that does have a C compiler available. If this is still an issue, please request a reopen of the ticket.

closed 2017-03-08 by foosel