OctoPrint/OctoPrint #1598: [BUG] Print gets on its own paused mid print without reason

ManuGithubSteam, 2016-11-24:
---

#### What were you doing?

Printing a gcode file with octoprint.

#### What did you expect to happen?

Print continues on its own as i did not press PAUSE button.

#### What happened instead?

Print paused several times without user interference. Had to Resume the print in the octo gui.

#### Branch & Commit or Version of OctoPrint

1.2.17 master

#### Printer model & used firmware incl. version

Prusa i3 mk2 lastes firmware 3.09

#### Link to octoprint.log

http://pastebin.com/kt4z48hY

#### Link to contents of terminal tab or serial.log

Send: N18698 G1 X152.607 Y116.128 E0.18703*107
Recv: ok
Send: N18699 G1 X153.803 Y113.487 E0.09227*103
Changing monitoring state from 'Printing' to 'Paused'
Recv: ok

(Hope) this is the relevant snippet. messages get cut of in the terminal window :-( Serial log will be submitet at the next print. maybe later tonight or tomorrrow. 

thanks for the help :) Have a nice day!

foosel, 2016-11-24:
```
2016-11-24 09:41:56,196 - octoprint.plugins.filament - INFO - Printing started. Filament sensor enabled.
2016-11-24 10:05:35,288 - octoprint.plugins.filament - DEBUG - Detected sensor [14] state [0]? !
2016-11-24 10:05:35,289 - octoprint.plugins.filament - DEBUG - Sensor [0]!
2016-11-24 10:08:12,028 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.249
2016-11-24 10:17:05,841 - octoprint.plugins.filament - DEBUG - Detected sensor [14] state [0]? !
2016-11-24 10:17:05,842 - octoprint.plugins.filament - DEBUG - Sensor [0]!
[...]
2016-11-24 10:27:35,620 - octoprint.plugins.filament - DEBUG - Detected sensor [14] state [0]? !
2016-11-24 10:27:35,621 - octoprint.plugins.filament - DEBUG - Sensor [0]!
2016-11-24 10:27:37,239 - octoprint.plugins.filament - DEBUG - Detected sensor [14] state [0]? !
2016-11-24 10:27:37,240 - octoprint.plugins.filament - DEBUG - Sensor [0]!
```

Please try without the filament sensor plugin. My guess is that this is misfiring here and pausing your prints for you. In general, first try to reproduce a bug without *any* third party plugins being enabled (= not marked as "bundled"). If it doesn't occur then, it's a plugin at fault.

PS: do not cut anything off the template or the bot will come, as just proven ;)

foosel, 2016-11-25:
@ManuGithubSteam had a chance to test this yet? I'm kinda curious ;)

ManuGithubSteam, 2016-11-27:
@foosel 
I just finished an 11 hour print without issue to i guess it is indeed filament plugin related. I will get in touch with the plugin dev soon.
Thanks for pointing that out and have a nice week!

closed 2016-11-27 by ManuGithubSteam