OctoPrint/OctoPrint #1599: Upgrade from 1.3.0rc1 to 1.3.0rc2 fails

JohnOCFII, 2016-11-25:
----

#### What were you doing?

Using the OctoPrint Software Update tool to upgrade from 1.3.0rc1 to 1.3.0rc2.

#### What did you expect to happen?

Expected upgrade to complete successfully.

#### What happened instead?

Upgrade appeared to be nearly complete, then failed.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0rc1 (rc/devel branch)

#### Printer model & used firmware incl. version

Original Prusa ik3 MK2 Firmware 3.0.9

#### Browser and Version of Browser, Operating System running Browser

Safari 10.0.1 MacOS Sierra 10.12.1

#### Link to octoprint.log

https://gist.github.com/JohnOCFII/d5bfa6b1be961dc0fa48556f5329e7e6.js

#### Link to contents of terminal tab or serial.log

"https://gist.github.com/JohnOCFII/fbfa2120c39cf149e763ac0de907490b.js

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2016-11-25:
Something apparently went wrong while downgrading your tornado version. Which I'll have to investigate. Thanks for reporting!

foosel, 2016-11-25:
Can you tell me what ``~/oprint/bin/pip freeze | grep tornado`` is currently reporting on that affected system? I cannot seem to reproduce the underlying error at the moment with my setup.

foosel, 2016-11-25:
Ah, never mind, I can reproduce it now.

foosel, 2016-11-25:
It looks like this is actually a duplicate/different manifestation of #1586, caused by a unicode character in the tornado downgrade output. Which means it's already fixed, but of course coming from 1.3.0rc1 you don't have the fix yet.

In my own reproduction it actually did update fine even though it reported the error, and a manual restart of the server after that had it running 1.3.0rc2.

Still, there's still an issue in there that I'd  like to see fixed (the error message you received hints at a bug in the update script caused by the other bug at that exact point in the installation), I'm not yet sure though it actually merits a new RC just for that.

I'll definitely add a note to the release notes and the release announcement on the blog though that this might happen under specific circumstances when upgrading from 1.3.0rc1 (and only then).

foosel, 2016-11-25:
So that specific error that was a consequence of #1586 and which you ran into there should also now be fixed with 528e99898fee11a5371451f8bb1dafae20be1bb8. I've pushed that to `devel`. If there will be a 1.3.0rc3 it will be part of that, otherwise it will be released with 1.3.1.

JohnOCFII, 2016-11-25:
Thank you for the quick investigation.  I had not even realized that the upgrade had, in fact succeeded until I read your note above.  I'll be sure to test out a few new features (bounding box and ignoring Z-hop in the timelapse) today.  Should I close this, or did you want to keep it open to track items mentioned in your note?

foosel, 2016-11-25:
Since there was something I had to do after all (getting rid of that follow-up error), I'd like to keep that open but marked as solved until the next version fixing THAT has been rolled out too.

Basically as a sort of book keeping, plus making already reported issues more visible.

foosel, 2017-01-25:
This should have been closed a while ago actually ;)

closed 2017-01-25 by foosel