OctoPrint/OctoPrint #1602: [Feature Request] Make the controls in the Control tab hide-able

ManuGithubSteam, 2016-11-25:
Hi fossel,
I would like to have the option that i can hide the control tabs in the Control section (with the webcam feed), even if i'm logged in. 

I know that if i log out, i will not see the controls, However i would like to stay logged in and minimize the chance to click something on the controls.

Thanks!

foosel, 2016-11-25:
Can you outline the usecase where that would be required? The actually critical controls are automatically disabled while printing to prevent accidents from happening.

foosel, 2016-11-25:
also, it's "foosel" :P

ManuGithubSteam, 2016-11-25:
Well when i access the octo gui with my mobile phone then everything is squished and tiny. I do not want to risk touching anything control related at all, even just the flow control. However this happend before and i switched the fan off. Due to internet lag i was not able to switch it back on again in time and the print was damaged. This could be avoided when the whole controls are not visible :-)

foosel, 2016-11-25:
Or by installing the TouchUI plugin and have a UI optimized for your phone ;)

ManuGithubSteam, 2016-11-25:
Ok i did not try that because it looked kinds ugly in the screenshot and i like the clean octo gui. However i will give it a try. Thanks

ntoff, 2016-11-25:
I'd just like to be able to hide them on my tablet when I have a bunch of custom controls so I don't have to scroll down to get to the custom controls, then back up to see the webcam. My tablet is big enough to not need  the touchUI plugin (which is  incompatible with screensquish anyway so I can't use it), but small enough that collapsible controls would be handy.

ManuGithubSteam, 2016-11-25:
I second ntoff, on this i did not even thought about my tablet *o_O*

MoonshineSG, 2016-11-27:
Or you can use the Nautilus plugin on you mobile (assuming is iOS) :)

donovan6000, 2016-11-27:
There's also the [M33 Fio](https://github.com/donovan6000/M33-Fio) plugin that makes all the control groups collapsible.
[image]

ntoff, 2016-11-27:
@donovan6000 unfortunately that doesn't work with the fan slider I've been trying to add to octoprint so isn't an option for me either. Plus I get a tonne of errors in it, something about null not having a remove property or something and *all* the buttons show up whether I have an m33 printer or not.

donovan6000, 2016-11-27:
What web browser are you using? And if you could create an issue over at [M33 Fio's page](https://github.com/donovan6000/M33-Fio/issues) with the errors that you get that we'd be more than happy to fix them :)

ntoff, 2016-11-27:
Chrome and my own abomination of octoprint https://github.com/ntoff/OctoPrint/tree/dev/fanspeedslider with a fan speed slider as an option. The trouble is with the slider enabled m33 fio doesn't know what to do with it so there's just a random slider left there that does nothing. As for the other errors, that'll have to wait until I can be bothered turning on my dev pc and I suspect they're probably more to do with the fact that I've modified octoprint than an actual error with the plugin (which is why I haven't bothered to create an issue ticket in the first place) so the errors are probably unique to my situation and not something that needs fixing. Still, it means I can't use that plugin.

donovan6000, 2016-11-27:
@ntoff Have you considered implementing the fan controls as [Octoprint custom controls](https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls)? They are much more easy to work with.

You can create similar fan controls by just adding the following to the end of your `~/.octoprint/config.yaml` file.
```
controls:
- children:
  - command: M106 S%(speed)s
    input:
    - default: 255
      name: Speed (0-255)
      parameter: speed
      slider:
        max: 255
        min: 0
    name: Enable Fan
  - command: M107
    name: Disable Fan
  layout: horizontal
  name: Fan
```
[image]

Also, sorry @foosel for getting this discussion so off topic ;)

ntoff, 2016-11-27:
I know you can add them there, the slider feature was sparked by a request for it on google+. Either way if you want to talk more about it, feel free to do it in the brainstorming thread https://github.com/foosel/OctoPrint/issues/1592

ManuGithubSteam, 2016-11-27:
@donovan6000 
Will the fifo plug in introduce other stuff that i probably don't want to use ? More precisely, can i disable every other feature in the fifo plugin other than the collapsing controls? To be honest i don't like the idea to install such a big plugin when i only want the controls hidden. But i would do it, if i can deactivate everything on it that is not useful for me (for now thats everything else than the control hiding)

Also i would like to have maybe one hide bar and not several for movement etc. Could this be implemented, so i can hide the complete controls at once?

Important is also, does it save the collapsed state somewhere or will i have to do it every time i log in into octo? This is even more important if i have to hide every section of it separately.

Will the webfeed still be shown ?
Thanks

foosel, 2016-11-27:
@donovan6000 No harm done, and I was actually about to tag you in #1592 for the issue @ntoff saw with his modifications anyhow.

donovan6000, 2016-11-27:
@ManuGithubSteam The only things that can be disabled in M33 Fio are the Micro 3D printer specific features, so you wont be able to disable all the other features. There's no big hide bar, and I'm not really sure If I'd want to implement one. I never made the webcam stream collapsible, so that'll always be shown. Here's what the controls tab looks like when there's a webcam stream and the Micro 3D printer specific features are disabled.
[image]
The collapsed state of each group is stored in the web browser's local storage, so their states are persistent.

LazeMSS, 2021-03-31:
Many plugins available for this:
https://plugins.octoprint.org/plugins/themeify/
https://plugins.octoprint.org/plugins/customcss/
https://plugins.octoprint.org/plugins/uicustomizer/