OctoPrint/OctoPrint #1604: [Brainstorming] Any way to detect Sailfish firmware and notify user if GPX plugin is missing?

foosel, 2016-11-28:
This is mostly a brainstorming ticket for @markwal :) Basically what the title says. OctoPrint in 1.3.0 now has firmware detection capabilities, but that currently only works through `M115`. I've been wondering for a while now if it was possible somehow to detect from what Sailfish returns on an attempt to connect with regular GCODE that the printer needs your GPX plugin to work. 

I would have given this a test myself but I still don't have access to a sailfish machine to test. Is there anything that would be usable for something like that, does Sailfish maybe send back some kind of error when it receives a handshake attempt from OctoPrint that we could use here to tell the user to please download your plugin?

markwal, 2016-11-29:
I'd thought about this a while back and investigated how the abandoned ReplicatorG (MakerBot opensource from before the about face) did detection.  They actually use the USB device ids.  On a linux system it shows up distinctly in /dev/serial/by-id.

However, to do it as exception handling when an unexpected response comes back from the serial port, Sailfish should always reply with the invalid checksum error code (single byte: 0x83) (unless we got really unlucky and the packet was accidentally well formed) or a generic packet error (0x80).  It'll probably return a lot of them actually.  I could try it out and see.  I wonder if we'll get false positives on just a bad or wrong speed connection with a regular Marlin printer though?

foosel, 2016-12-01:
Hm, yeah, false positives are definitely a possibility there if it's only a single byte. 

The USB device ID might be accessible through current PySerial (not the one OctoPrint uses, want to upgrade anyhow). Maybe that might be the better option then.

foosel, 2017-01-27:
Closing this for now, might get back to you at a later date once I've upgraded pyserial.

closed 2017-01-27 by foosel