OctoPrint/OctoPrint #1605: [Request]: Ability to disable OctoPrint port for Flask and Python usage

tanthua, 2016-11-28:
Hi everyone,
I'm currently building a new plugin for OctoPrint running on Rasberry Pi 3. What I want it to do is to have a HTML/jinja2 front-end tab that contains buttons to trigger a command to capture the image from RaspiCam, and another button to run a python script with OpenCV that will do edge detection and output to the front-end the image after processing. 

I can do all of those things manually but I have not found a way to automate it. I tried using `flask` as recommended after search but it ran into the "Address already in use" issue because OctoPrint is running on it. So, my question is if there is a way to do such things? In other words, is there a way to trigger HTML/jinja2 buttons to trigger python and bash script in the Raspberry Pi without having the port or socket issue when OctoPrint is still running?

The python code got from `flask`, which will be the python code with OpenCV imported and do image processing. This one was the on that output the "Address already in use" error.

    from flask import Flask
    app = Flask(__name__)
    
    @app.route("/")
    def hello():
        return "Hello World!"
    
    if __name__ == "__main__":
        app.run()

The HTML/jinja2, currently using the button to display date and time, but will trigger the python code above and a command line:

    <html>
    <body>
    
    <h1>Java Script</h1>
    
    <button type="button" onclick="document.getElementById('demo').innerHTML = Date()">
    Click me to display Date and Time.</button>
    
    <p id="demo"></p>
    
    </body>
    </html>

And my command line to capture picture from mpjg-streamer is:

    wget http://localhost:8080/?action=snapshot -O output.jpg

If possible, please provide me an example with explanation for it? Thanks. My apology if my post is not supposed to be here. I have checked and asked around and still not be able to find the solution for this.

markwal, 2016-12-01:
You can easily have buttons on the system menu that trigger command lines.  But since you asked for a tab and you want to output an image to the front end in the tab, you'll want to implement a plugin.  From the way you ask your questions, I'd recommend working through a few tutorials for extending OctoPrint first before coming back to your original goal.  Getting the tab to show up and executing the command lines will be fairly straightforward.  I suspect the difficult part will be coming up with a way to return the generated image to the client.  That is, having the image show up in the namespace in a reasonable way that you can reference from your HTML.

OctoPrint already uses flask, but you'll want to use OctoPrint's instantiation of flask, so the way to do this is as a TemplatePlugin into OctoPrint.

In any case, you'll want to work your way through the getting started tutorial: http://docs.octoprint.org/en/master/plugins/gettingstarted.html.   That tutorial includes added templated html (jinja) to the OctoPrint ui.

After that, if you want to add custom route's to the flask setup, then the BlueprintPlugin mixin will be interesting to you: http://docs.octoprint.org/en/master/plugins/mixins.html#blueprintplugin

tanthua, 2016-12-01:
Hi Mark, thank you for your response. I have gone through the tutorial and documentation several times and I got the idea. I'm currently sending direct command line to to  those works instead of buttons. So I'm happy with that for now.
Also, do you have an idea of how to link the local file into jinja? That's what I've been struggling with because I thought it should be simple. I have the file located at /static/images/image.jpg and my jinja2 code is:
`<img src="/static/images/image.jpg">`
or
`<img src={{ url_for('static',filename="image.jpg")}}>`
or
`<img src="/{{myplugin_url}}/static/images/image.jpg}}">`

And none of them was displayed. I'm wondering if I need to do any additional steps to get that linked together.

tanthua, 2016-12-01:
I figured the path out. It is in this `/plugin/my_plugin/static/images/image.jpg` format. Thanks for providing me the hint Mark. Now, I'm working on the image changed dynamically when another snapshot is taken.
I have the path stored in myplugin.js as:
`self.imagePath = /plugin/my_plugin/static/images/image.jpg`
and in the myplugin_tab.jinja2:
'[image]'
And the command line for snapshot is:
`wget http://localhost:8080/?action=snapshot -O /plugin/my_plugin/static/images/image.jpg`

But the image is not changed every time I take a new snapshot even after reloading the page. I checked the file statistic, and it does show it is overwritten by the new image.jpg.

My apology for asking too many questions in one issue thread. Thanks a lot!

markwal, 2016-12-01:
It's probably server side caching that is causing that problem.  It's probably not a good idea to have "static" folders contain dynamic data. Especially since when it is installed it'll be in the python module library.  I'm thinking you should store it in the plugin data folder under .octoprint and then figure out how to return the content via flask or tornado maybe.

tanthua, 2016-12-02:
I see. Thank you so much for your help!

closed 2016-12-02 by tanthua