OctoPrint/OctoPrint #1607: [Request] 1Password compability

ss89, 2016-11-30:
Hello developer(s),

i have found an annoying "bug" (well not really a bug but an incompatibility) which bothers me for quite some time now.
Since i am getting logged out quite often, which is ok for security reasons, i have to login quite often.
Usually i do this on websites via 1Password which is quite fast since i have my 1Password keychain unlocked like always when my computer is in active use because i use it so much.
Doing this is just like: clicking the 1Password Browser Extensions icon, click on the entry for the octoprint website, then it (should) fill it out (which it doesn't, but works on really every other website).
I believe it might have to do how octoprint handles logins, since the login popover is being closed after the login (would have been) filled out.
If you could make this, that would be so great.
@foosel will also give you a purchased license for the platform of your choice (be it windows or mac).

best regards,
ss89

ctgreybeard, 2016-11-30:
I second this request. I was going to look into why 1Password fails to detect the login form but haven't gotten around to looking at it.

numanair, 2016-12-04:
With the chrome extension for KeePass I have to redetect the login field after it pops up. An upcoming release will support plugins that can force login to access any part of the interface. I believe that such a plugin would also solve this.

BillyBlaze, 2016-12-05:
I have added some basic support to support all password managers, sadly due to the conditions made by browser this will not always prompt in Chrome or 1Password due to the fact that they don't support saving from a non-SSL websites.

This implementation should work if they do add support in the future for non-SSL websites.

foosel, 2016-12-05:
And just for the record: It's not possible to have OctoPrint listen on https out of the box with a certificate that will not make your browsers scream bloody murder, especially not for internal LAN-only hosts (e.g. octopi.local). Not because of OctoPrint but because of how the whole SSL/https infrastructure works.

BillyBlaze, 2016-12-13:
Small update, my commit has been merged to maintenance (1.3.1)

After another test, Chrome will now also prompt their built-in password manager after submitting the form.

I still couldn't test this with 1Password 6. But we should've everything in place to make this work with any password manager.

Can someone confirm that this also works with 1Password 6?

ss89, 2016-12-30:
I can not confirm that this works on 1Password 6, i just tried it in chrome.
Neither am i asked to save the login (1Password wise), nor will it login when create the login manually and then try to use it (also 1Password wise).
Please see this screenshot for how my login looks like (web formular details).
Is there anything missing that would be needed?
[image]

BillyBlaze, 2016-12-30:
can you change `login_user` and `login_pass` into `username` and `password`?

I tried this with 1password 4 and with Dashlane and both password managers where able to save the credentials.

ss89, 2016-12-31:
Thank you for pointing that out, ofcourse they should be named by the name attribute and not the id.
Sadly it still doesn't work for me.
What i did forget to mention: everytime after i tried it and reopened the 1password addon it wants to reestablish its connection to the 1password mini app.
Other Logins still work though.
@BillyBlaze could you possibly post a screenshot of what your web formular details look like?

BillyBlaze, 2016-12-31:
First off, I have made 2 extra commits (ce1344eb0ccf92fa79775e4b1e1efdca1883c62c, 6cc84a1473e3a8664defa6d23e77658919440c74) that haven't arrived yet on the maintenance branch, I first want @foosel to take a look at it.

However, can you checkout the branch ```main/supportPasswordManagers```?

After that, please remove your OctoPrint entry from 1Password, then open OctoPrint, click on 1password so that mini app is connencted to the main app, and then try to login with typed credentials.

Let's first try to see if this works in 1password 6, because if everyone needs to make manual forms I rather find out why this doesn't work automated 👍

ss89, 2016-12-31:
i have just checked out 6cc84a1473e3a8664defa6d23e77658919440c74 with version number "1.4.0.dev26+g6cc84a1 (main/supportPasswordManagers branch)" and restarted octoprint daemon.
I have removed my 1Password entry for that instance completely and don't see anything to auto fill out.
Then i tried to login, which succeeded, but 1Password doesn't ask for adding a new login.
Also after manually creating the login it won't fill out the form.
1Password Extension was connected to 1Password mini at that time.
Interesting how 1Passwords behavior seems to have "evolved" since Version 4 (i had the feeling that stuff got worse since Version 3, where i thing http auth was supported but got dropped later on).
Do you have any other recommendations on what to try? I usually don't bug much with login forms.

BillyBlaze, 2016-12-31:
Argh that is really a bummer 😞 I hoped it would behave better now...

Here is my webform settings: _(note: password is not real for anyone wondering)_
[image]

If manually also will not work, then I have no other ideas then to ask about this on the friendly 1password forum, it seems like we're now doing everything correctly for them to pick up the rest.

foosel, 2017-01-27:
As far as I understand, there's nothing we can do better now from our side thanks to @BillyBlaze. We are not hiding the login form any longer, the input fields are properly marked... All released with 1.3.1

@BillyBlaze can we close this from your point of view as well?

BillyBlaze, 2017-01-27:
Yes, we can imo. We're doing everything as specified.

foosel, 2017-01-27:
Ok, closing this then :)

closed 2017-01-27 by foosel