OctoPrint/OctoPrint #1610: updating to 1.2.18 from 1.2.17 fail

thomllama, 2016-11-30:
OK, ppped up and said "new update"  hit the download.. ask to proceed and click yes.. and instant fail to download message. think I remember in the past there was something in the settings to do before the update would take.. but can't find it and don't remember.. any ideas? 

[image]

thomllama, 2016-11-30:
uggg . I have NO clue about all that info.. it is running Octopi version 1.2.17 . update says there is a 1.2.18 . but wont work... simple question.

thomllama, 2016-11-30:
OK . found the old fix.. have to open the setting and just hit "save" for some reason with the last update.  just did it again and it worked.

thomllama, 2016-11-30:
OK.. now it says its's updating.. but fails to finish . :/

markwal, 2016-11-30:
Closing because we're tracking in the new bug, right?

closed 2016-11-30 by markwal

thomllama, 2016-11-30:
ya...  was trying to get all the info so it wouldn't come up "incomplete". didn't work here so I tried a new post.. we can delete this if you like