OctoPrint/OctoPrint #1611: 2nd try.. Update to 1.2.18 from 1.2.17 fail

thomllama, 2016-11-30:
OK.. trying again... got to say the "requirements are a bit much.. I understand the need to info but many people just don't have that much geek in them to understand or retrive a lot of it. 

**What were you doing?**

trying to update to 1.2.18 from 1.2.17

**What did you expect to happen?**

for it to update
**What happened instead?**
failure message(s)

**Branch & Commit or Version of OctoPrint**
1.2.17 (master branch)

Printer model & used firmware incl. version
Max Micron  - marlin version 1

#### Browser and Version of Browser, Operating System running Browser
Mac OSX Sierra  Chrome browser

 Link to octoprint.log
??????
[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

 Link to contents of terminal tab or serial.log
?????

Link to contents of Javascript console in the browser
??????

Screenshot(s) showing the problem:
![screen shot 2016-11-30 at 3 18 04 pm]

thomllama, 2016-11-30:
log- 2016-11-30 20:04:08,879 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'en', 'path': '/', 'base_url': 'http://octopi.local/'}
2016-11-30 20:04:10,376 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.144
2016-11-30 20:04:39,520 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-11-30 20:11:44,292 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-11-30 20:34:48,269 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-11-30 20:34:48,309 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.144
2016-11-30 20:34:49,033 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.144
2016-11-30 20:34:50,217 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.2s
2016-11-30 20:34:54,898 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.2.18...
2016-11-30 20:46:25,804 - octoprint.plugins.softwareupdate.updaters.update_script - ERROR - Target: octoprint, error while executing update script, got returncode 1
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 68, in perform_update
    returncode, stdout, stderr = execute(update_command, cwd=folder)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/util.py", line 25, in execute
    raise ScriptError(p.returncode, p.stdout.text, p.stderr.text)
ScriptError
2016-11-30 20:46:25,809 - octoprint.plugins.softwareupdate.updaters.update_script - WARNING - Target: octoprint, update stdout:
>>> Running: git diff --shortstat
>>> Running: git fetch

2016-11-30 20:46:25,810 - octoprint.plugins.softwareupdate.updaters.update_script - WARNING - Target: octoprint, update stderr:
error: RPC failed; result=56, HTTP code = 200
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 204, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 200, in main
    update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 109, in update_source
    raise RuntimeError("Could not update, \"git fetch\" failed with returncode %d: %s" % (returncode, stdout))
RuntimeError: Could not update, "git fetch" failed with returncode 128: 

2016-11-30 20:46:25,811 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 689, in _perform_update
    update_result = updater.perform_update(target, check, target_version)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 75, in perform_update
    raise UpdateError("Error while executing update script for %s", (e.stdout, e.stderr))
UpdateError
2016-11-30 20:59:39,766 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.2.18...
2016-11-30 21:12:56,083 - octoprint.plugins.softwareupdate.updaters.update_script - ERROR - Target: octoprint, error while executing update script, got returncode 1
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 68, in perform_update
    returncode, stdout, stderr = execute(update_command, cwd=folder)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/util.py", line 25, in execute
    raise ScriptError(p.returncode, p.stdout.text, p.stderr.text)
ScriptError
2016-11-30 21:12:56,085 - octoprint.plugins.softwareupdate.updaters.update_script - WARNING - Target: octoprint, update stdout:
>>> Running: git diff --shortstat
>>> Running: git fetch

2016-11-30 21:12:56,086 - octoprint.plugins.softwareupdate.updaters.update_script - WARNING - Target: octoprint, update stderr:
error: RPC failed; result=56, HTTP code = 200
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 204, in <module>
    main()
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 200, in main
    update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 109, in update_source
[... cut ...]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 689, in _perform_update
    update_result = updater.perform_update(target, check, target_version)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 75, in perform_update
    raise UpdateError("Error while executing update script for %s", (e.stdout, e.stderr))
UpdateError

markwal, 2016-11-30:
It looks like git (the source control tool the updater uses to get the source code for the version you're updating to is unable to actually fetch the new version) was unable to get the new version from github.  "fatal: The remote end hung up unexpectedly".  Your internet is mostly working for the pi since it's able to connect and get a result back, but it isn't able to get all the data it needs to actually perform the update.

It could have been that github.com was having a bad day maybe?  Or perhaps the pi's access to the internet is tenuous?

Normally, if github is down, it fails earlier and I suspect the error message is a bit more understandable.

In your shoes (though this will require some geek), I'd try git fetch from the command line. Since it appears you are running from octopi image, it'd be like the following.  If octopi.local access by name works for you, you can use that instead of the ip address.
```
ssh pi@<octopi_ip_address_here>
source ~/oprint/bin/activate
cd ~/OctoPrint-GPX
git fetch
```

If that appears to work, you can try the update from the web again since it'll already have the data it needs from github.  If it doesn't, I'd be interested in the verbatim of the error messages (even if it is the same as above).

markwal, 2016-11-30:
Watch out.  The first time I typed the command lines, I had a typo, so you'll want to go off the web version not the email (if you get notifications).  Also, those tilde (~) are important.

thomllama, 2016-11-30:
yaaaa.....aaaaaa. not so much a unix/linux geek.. can often figure it out,.. but,  kinda a hair puller for me.  printer is printing right now so .. it'll have to wait.  I'll try tomorrow if I get time after work.  

thanks though!!

foosel, 2016-12-01:
> I understand the need to info but many people just don't have that much geek in them to understand or retrive a lot of it.

The [Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md) linked on top of every ticket submission form, within the default ticket text and by the bot actually are supposed to help with that, explaining where to find the stuff that's asked for.

If you read those and things were still unclear, I'd like to understand which parts so that can be improved :)

thomllama, 2016-12-08:
Still having update issues.. its now 1.3.0. takes like 20 minutes to come up and say it fails... :/ 

will paste in log in next comment- 

as to the confusing Sumitting form.. it's just text.. not really sure what's needed.. do I keep the form as is and fill n/add info?  do I replace each line with the info?  getting/linking to like what 3-4 other uploads sites/locations is rather confusing at best.  what not have an actual "form" with input areas and such... 1/2 the time I'm not sure if I should delete part and replace or just fill in or what..

thomllama, 2016-12-08:
2016-12-02 02:06:26,425 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Pi3_tree_v1.gcode
2016-12-02 02:13:42,286 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.144
2016-12-02 02:29:59,129 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.144
2016-12-02 02:30:00,770 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.58s
2016-12-02 02:30:18,468 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2016-12-02 02:31:39,533 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.144
2016-12-02 02:17:08,489 - octoprint.server - INFO - Starting OctoPrint 1.2.17 (master branch)
2016-12-02 02:17:08,556 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins and installed plugin packages...
2016-12-02 02:17:09,247 - octoprint.plugin.core - INFO - Found 6 plugin(s) providing 6 mixin implementations, 3 hook handlers
2016-12-02 02:17:09,252 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2016-12-02 02:17:09,273 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2016-12-02 02:17:09,293 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2016-12-02 02:17:11,367 - octoprint.util.pip - INFO - Found pip at /home/pi/oprint/bin/pip, version is 8.1.1
2016-12-02 02:17:11,369 - octoprint.plugin.core - INFO - Initialized 6 plugin implementation(s)
2016-12-02 02:17:11,372 - octoprint.plugin.core - INFO - 6 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/announcements
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/virtual_printer
2016-12-02 02:17:11,373 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2016-12-02 02:17:11,378 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2016-12-02 02:17:11,389 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2016-12-02 02:17:11,399 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2016-12-02 02:17:12,116 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _http._tcp
2016-12-02 02:17:12,130 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for _octoprint._tcp
2016-12-02 02:17:12,133 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2016-12-02 02:17:12,484 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from http://plugins.octoprint.org/plugins.json
2016-12-02 02:17:12,539 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2016-12-02 02:17:12,624 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://octopi.local/'}
2016-12-02 10:56:04,485 - octoprint.server - INFO - Preemptively caching / for {'query_string': 'en', 'path': '/', 'base_url': 'http://octopi.local/'}
2016-12-02 10:58:18,128 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.144
2016-12-02 11:01:15,157 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.144
2016-12-02 11:12:03,301 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.144
2016-12-02 11:12:04,499 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.22s
2016-12-02 11:12:12,002 - octoprint.filemanager.analysis - INFO - Starting analysis of local:test_box.gcode
2016-12-02 11:19:53,956 - octoprint.filemanager.analysis - INFO - Starting analysis of local:test_box.gcode
2016-12-02 11:23:59,701 - octoprint.filemanager.analysis - INFO - Starting analysis of local:test_box.gcode
2016-12-02 11:27:56,975 - octoprint.filemanager.analysis - INFO - Starting analysis of local:test_box.gcode
2016-12-02 11:30:27,395 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.144
2016-12-02 11:33:00,726 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.144
                                                                                                     2016-12-02 11:17:08,450 - octoprint.server - INFO - Starting OctoPrint 1.2.17 (master branch)
2016-12-02 11:17:08,515 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins and installed plugin packages...
[... cut ...]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 689, in _perform_update
    update_result = updater.perform_update(target, check, target_version)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.17-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 75, in perform_update
    raise UpdateError("Error while executing update script for %s", (e.stdout, e.stderr))
UpdateError

foosel, 2016-12-08:
That's still the same issue, your Pi cannot talk reliably with the update server:

```
error: RPC failed; result=56, HTTP code = 200
fatal: The remote end hung up unexpectedly
```

Something is up with your Pi's network connection to Github (where the update is supposed to be retrieved from). If your Pi is connected via Wifi, could you maybe try a cable, just to rule out a shaky network as the culprit?

---

Now about your issues with the bug reporting process, you said:

> it's just text.. not really sure what's needed.. 

The text literally says

> If not already done, please read the "guidelines for contributing"
> that are linked ^-- just up there in the big yellow box.

And exactly above that text is a big yellow box with a link (one link I might add, not 3-4) to click and read. And that link has pictures and explains where to find all the information (like logs and version numbers and such) that are asked for. I have no way to make that better than that because:

> what not have an actual "form" with input areas and such

Github sadly does not allow me that degree of customization for the bug tracker, or I would do that, trust me - it would also save ME a lot of work hunting down after information in incomplete tickets ;) And hosting a different tracker myself would add a lot of system administration overhead that would then steal time from actual development. And having it hosted would cost money, and might still not allow the necessary amount customization.

> 1/2 the time I'm not sure if I should delete part and replace or just fill in or what

Quoting again from the text:

> Do not delete any lines from the template but those enclosed in [ and ] - and those please DO delete, they are only provided for your information and removing them makes your ticket more readable :)

I admit it's necessary to read a bit in order to know what to do when reporting an issue, but the information should all be there. And considering that as an open source project OctoPrint can't really afford manning a full fledged support desk 24/7 to take care of these kinds of questions, offering the information in textual form is sadly the only way to handle this in a viable way that also leaves time for the actual development :/

It's basically part of the cost of having this project be free and open.

thomllama, 2016-12-08:
oh man it's info over load for me..  I have no clue how you can deal with all of it.. and then people like me ;). 

OK. to the main issue.. the Pi Is hardwired to the modem.. so connection on my end isn't an issue in the sense of physical connection..   is there a way I can just download it to my computer an have the pi install it from there?

foosel, 2016-12-08:
Not without having to "geek" a bit I fear. Firing up Putty, logging in via SSH and executing some commands through that will be involved. And then what @markwal suggested up there would actually be the easier way.

thomllama, 2016-12-08:
and what I meant by multiple links or servers or what ever is here... 
--------------
#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### **Link to contents of terminal tab or serial.log**

**[On gist.github.com or pastebin.com.** If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### **Link to contents of Javascript console in the browser**

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

thomllama, 2016-12-08:
Not trying to be a pain here.. know you guys work REALLY hard on this stuff.. but would be nice to have a way to just download the updates and install locally instead of having to try and fetch it and install over the net... just a thought...?  especially for us with lower geek ability :P

foosel, 2016-12-08:
> [links]

That's where you are supposed to paste that stuff so it becomes easier readable than just pasting thousands of lines of code into a comment here (and with less risk of stuff being mangled in the process). And it's either pastebin.com or gist.github.com, offered as alternatives. This is again in order to make ticket handling and identifying support vs actual issues easier and less time consuming - not because I'm lazy but because this kind of stuff can eat an extraordinary amount of time that is then missing for doing actual development work for everyone. Again, trade offs of an open source project - the users are expected to help where they can to keep stuff running (be it by something as simple saving the devs time spent on running after ticket info) and in turn the project is open and free.

> [allowing to have downloadable packages]

That would make it even more complex to have it stay updated without much user interaction (the geek part). If you weren't running into those network issues, be honest, what would you rather have? Dialog box with button "Update now" or "now available, download package here"? Both is not possible to keep working regardless of with what you started with, not for a headless (no monitor) server side program, not without a huge overhead for maintenance and even more debugging of semi-broken setups. This doesn't scale, not with the available resources (read: 1 person working full time on it and a handful of people putting in time when they can). Having the current automatic update in place was already tricky enough to get to work reliably across the very very fragmented landscape of possible setup scenarios.

thomllama, 2016-12-08:
same issue.. seems it connects.. then it just drops off to zero transfers .. :/ 

[image]

markwal, 2016-12-08:
> hmm is the Octopi install updated? could I just download and and re do a whole new card with everything up to date?

Official releases of the OctoPi image are much less frequent than of OctoPrint.  It's quite a bit bigger and for most, a heavier hammer for upgrade (you don't keep your settings when you re-image unless you geek a bunch).  Plus OctoPi has a bunch of other components that all get validated together which takes longer.

markwal, 2016-12-08:
That's a pretty tenuous connection and small pipe.  Do you see it being this slow when you're just accessing the OctoPrint in the web browser?

thomllama, 2016-12-08:
nope.. the pi is almost instant reaction with anything I input in the browser...   both the compute and the Pi are cables to the modem. which is brand new... bot then pi it's self and the modem (is a Comcast/Xfinity modem..that might be an issue?  ;) )   funny it's still saying 0 bytes/s  transferring but it just jumped to 35% done

thomllama, 2016-12-08:
OK, just jumped to 65%.   seems it will work though may take for ever.. but .. when this is done.. it's done? installed? just like restart the pi?  or do I need to do some install commands?

markwal, 2016-12-08:
Weird.  I wonder why the comcast router is throttling the pi so much.  Or maybe it's github?  You could clone the repo onto your Mac for grins to do a speed test there.

thomllama, 2016-12-08:
not a clue what you just said. ;). LOL

thomllama, 2016-12-08:
[image]

markwal, 2016-12-08:
When it's done, then my guess is that the Update Now button in OctoPrint will work because the data is already local.  It'll still do a "git fetch" and talk to github, but that shouldn't fail because not many bytes are required to say: you've already got the latest, kbye.

But I don't know, at that trickle rate, I'd be surprised if it completes successfully.

The pi still has plenty of local storage available, right?

thomllama, 2016-12-08:
ya.. it's a 16gig card..

thomllama, 2016-12-08:
WOOHOOO 73%. LOL.  

Hey.. I know you guys have lives and I can be annoying !! appreciate you time/help... hey On that note.. Kinda poor but would love to donate a few bucks.. is there a place to do that?  think I looked back when I 1st started and it seemed like it was a monthly thing.. which I can't do.. need a 1 time shot at the moment...

thomllama, 2016-12-08:
well, it failed... :( 

[image]

thomllama, 2016-12-08:
just restarted.. same thing. gets to that 9% and hangs.. :/
[image]

thomllama, 2016-12-08:
OK, restarted Pi, restarted Modem/router... restarted computer.. same thing. gets to the 9% real fast then chokes... guessing you guys have not a clue anymore than I do?

markwal, 2016-12-08:
In your shoes, I'd find a way to hard wire that pi into the network at least for update purposes.

Failing that, you could download the update onto your Mac and copy it from there, but to do that you'll need some more geek:

1. Install git
2. Install scp
3. clone the repositiory onto your mac.  In the Mac terminal:
```
git clone https://github.com/foosel/OctoPrint.git
```
4. copy that repository from the Mac to the Pi
```
scp -rp OctoPrint pi@octopi.local:~
```

I'm a little worried about that last bit.  Hopefully that sets the owner of those new files on the pi to "pi", but I'd have to try it to know.  :-)  (`ls -la OctoPrint` on the pi would tell you)

markwal, 2016-12-08:
I use homebrew to install command line tools, but I think git is available to install on the Mac other ways and scp might come with the terminal utilities already, I'm not sure.

thomllama, 2016-12-08:
PI is hardwired.. Ethernet cable direct to modem/router .. reason this is bothering me so much

markwal, 2016-12-08:
git for Mac: https://git-scm.com/download
scp apparently already comes on the mac (you can type scp and hit return in the terminal to check)

thomllama, 2016-12-08:
OK.. well it runs on the current version.. my super geek Computer Scientist Brother should be down for Xmas (yes, he really is a computer scientist LOL that's his job title at Redhat) bet he can figure it out in 5 minutes.. I'm guessing it has to do wit the whole Comcast modem/router/firewalls/software BS

markwal, 2016-12-08:
OK.  Then I propose we close this github issue since we haven't identified anything to fix in OctoPrint.  Since the issue shows up without OctoPrint involved at all, it's unlikely to be an OctoPrint specific issue.  Feel free to post follow up questions and support requests on the Google Group and or Google+ Community.  Also, please do let us know if your brother figures out how to get it unstuck and what he did in case someone else hits the same problem.

closed 2016-12-08 by markwal

thomllama, 2016-12-08:
K.. been on the phone with Comcast for the last hour.. shut down all firewall and such items.. did all kinds of tracing and speed tests.. nutting. 👎  moved it around to next to the router/modem.. new ethernet cable shut wifi down completely.. only thing on my network is his computer and the Pi.. still nothing.. guess Git hub doesn't like my Pi. :/

oh well

markwal, 2016-12-08:
Huh.  I am really curious to see if your Mac can download from github ok.  You could try just downloading the .zip of the sources for 1.3.0 here:   https://github.com/foosel/OctoPrint/archive/1.3.0.zip and see if the bandwidth works for that.

thomllama, 2016-12-08:
clicked the link you listed.. downloaded in like 2 seconds 
[image]

thomllama, 2016-12-08:
can I just load that onto the SD card directly?  that would be sooo nice

foosel, 2016-12-08:
Without having tried that, and with the caveat that this will not allow you to update through the web interface and that you might run into problems later (because again this is untested and considering the hour I'm also not sure I can properly overlook potential issues with it at a later date):

Put that zip on your pi: `scp OctoPrint-1.3.0.zip pi@octopi.local:`

Ssh into your pi: `ssh pi@octopi.local`

Install 1.3.0 from that zip: `~/oprint/bin/pip install OctoPrint-1.3.0.zip`

Restart OctoPrint: `sudo service octoprint restart`

foosel, 2016-12-08:
But really, no guarantees!

thomllama, 2016-12-08:
cool.. thanks. that looks like I something I might be able to accomplish!!   you guys are awesome!!   might want to add the ability to local download/update in future. ;). LOL. Ill let you know if it works.. but otherwise.. have a good night. :)

thomllama, 2016-12-08:
well I give up.. tried your listed method.. seems close.. probably something I'm doing/not doing..  guy on facebook page trying to help also... still nothing.. same errors... here's what I got if it helps you down the road at all... 

Last login: Thu Dec  8 17:32:58 on ttys000
Ds-Mac-mini:~ d.thomaslamourine$ scp
usage: scp [-12346BCpqrv] [-c cipher] [-F ssh_config] [-i identity_file]
           [-l limit] [-o ssh_option] [-P port] [-S program]
           [[user@]host1:]file1 ... [[user@]host2:]file2
Ds-Mac-mini:~ d.thomaslamourine$ scp OctoPrint-1.3.0.zip pi@octopi.local:
pi@octopi.local's password: 
OctoPrint-1.3.0.zip: No such file or directory
Ds-Mac-mini:~ d.thomaslamourine$ ssh pi@octopi.local
pi@octopi.local's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Dec  8 22:33:18 2016 from 10.0.0.144
pi@octopi:~ $  ~/oprint/bin/ directory
-bash: /home/pi/oprint/bin/: Is a directory
pi@octopi:~ $ cd oprint/bin/
pi@octopi:~/oprint/bin $ ~/ pip install OctoPrint-1.3.0.zip
-bash: /home/pi/: Is a directory
pi@octopi:~/oprint/bin $ pip install OctoPrint-1.3.0.zip.
-bash: pip: command not found
pi@octopi:~/oprint/bin $ pip install OctoPrint-1.3.0.zip
-bash: pip: command not found
pi@octopi:~/oprint/bin $ ls
activate          ndg_httpclient  pyrsa-decrypt          python
activate.csh      netaddr         pyrsa-decrypt-bigfile  python2
activate.fish     octoprint       pyrsa-encrypt          python2.7
activate_this.py  pip             pyrsa-encrypt-bigfile  unidecode
easy_install      pip2            pyrsa-keygen           watchmedo
easy_install-2.7  pip2.7          pyrsa-priv2pub         webassets
markdown_py       pkginfo         pyrsa-sign
miniterm.py       pybabel         pyrsa-verify
pi@octopi:~/oprint/bin $ ~/oprint/bin/pip install OctoPrint-1.3.0.zip
Requirement 'OctoPrint-1.3.0.zip' looks like a filename, but the file does not exist
Processing ./OctoPrint-1.3.0.zip
Exception:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 209, in main
    status = self.run(options, args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 299, in run
    requirement_set.prepare_files(finder)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 360, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 577, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 798, in unpack_url
    unpack_file_url(link, location, download_dir, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 705, in unpack_file_url
    unpack_file(from_path, location, content_type, link)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 599, in unpack_file
    flatten=not filename.endswith('.whl')
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 482, in unzip_file
    zipfp = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/oprint/bin/OctoPrint-1.3.0.zip'
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
pi@octopi:~/oprint/bin $ pip install 'OctoPrint-1.3.0.zip'
-bash: pip: command not found
pi@octopi:~/oprint/bin $ pip install 'OctoPrint-1.3.0.zip'
-bash: pip: command not found
pi@octopi:~/oprint/bin $

markwal, 2016-12-09:
You've got to be in the folder with the zip file (or give its full path):

In the mac terminal before scp do this:
```
cd ~/Downloads
ls OctoPrint*
```

That should put you in the downloads folder and list out the zip file.  Then you can try the scp line: `scp OctoPrint-1.3.0.zip pi@octopi.local`

Don't go on with the rest unless you can get that to actually copy.

markwal, 2016-12-09:
Also, Linux doesn't include the current directory when looking for commands, so cd oprint/bin doesn't help it find pip. Type the whole thing out:

~/oprint/bin/pip install OctoPrint-1.3.0.zip

Besides you need to stay in the directory that contains the zip so that the pip can find it since that isn't a full path.

thomllama, 2016-12-09:
still nope... 

Last login: Thu Dec  8 18:51:55 on ttys000
Ds-Mac-mini:~ d.thomaslamourine$ scp cd ~/Downloads
cp: cd: No such file or directory
Ds-Mac-mini:~ d.thomaslamourine$ cd ~/Downloads
Ds-Mac-mini:Downloads d.thomaslamourine$ ls OctoPrint*
OctoPrint-1.3.0.zip
Ds-Mac-mini:Downloads d.thomaslamourine$ scp OctoPrint-1.3.0.zip pi@octopi.local
Ds-Mac-mini:Downloads d.thomaslamourine$ ssh pi@octopi.local
pi@octopi.local's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Dec  8 23:52:50 2016 from 10.0.0.144
pi@octopi:~ $ ~/oprint/bin/pip install OctoPrint-1.3.0.zip

Requirement 'OctoPrint-1.3.0.zip' looks like a filename, but the file does not exist
Processing ./OctoPrint-1.3.0.zip
Exception:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 209, in main
    status = self.run(options, args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 299, in run
    requirement_set.prepare_files(finder)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 360, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 577, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 798, in unpack_url
    unpack_file_url(link, location, download_dir, hashes=hashes)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 705, in unpack_file_url
    unpack_file(from_path, location, content_type, link)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 599, in unpack_file
    flatten=not filename.endswith('.whl')
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 482, in unzip_file
    zipfp = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: '/home/pi/OctoPrint-1.3.0.zip'
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
pi@octopi:~ $ 
pi@octopi:~ $ 
pi@octopi:~ $

thomllama, 2016-12-09:
and we were trying with the file directly dragged onto the pi's Boot drive using the Mac.. which sorta worked.. the file is there.. on the SD card... but... who knows...

foosel, 2016-12-09:
Please make sure you use the *exact* commands provided. Your scp command in your final attempt up there for example is missing the : at the end which is crucial for the command to do what you need it to do here (copy the file to your pi and there into your home directory instead of copying it into another file on your Mac called pi@octopi local). And I just saw that @markwal forget the : too up there, tsk - you may blame him ;)

You need to run these exact commands (do not leave out any characters, do not substitute anything or try to split commands, double check you typed them in correctly, when you get an error stop and check again):

Switch to the download folder on your Mac: `cd ~/Downloads`

Copy the file to the pi: `scp OctoPrint-1.3.0.zip pi@octopi.local:`

Ssh into the pi: `ssh pi@octopi.local`

Install: `~/oprint/bin/pip install OctoPrint-1.3.0.zip`

Restart: `sudo service octoprint restart`

thomllama, 2016-12-09:
HOLY CRAP!!!  it worked!!!!  :) :) :) 

thanks SOOOOOO much .. 

Last login: Fri Dec  9 05:03:39 on ttys000
Ds-Mac-mini:~ d.thomaslamourine$ cd ~/Downloads
Ds-Mac-mini:Downloads d.thomaslamourine$ scp OctoPrint-1.3.0.zip pi@octopi.local:
Warning: Permanently added the ECDSA host key for IP address '2601:188::75ef:627f:a964:fddf:cebe' to the list of known hosts.
pi@octopi.local's password: 
OctoPrint-1.3.0.zip                           100% 3675KB   3.6MB/s   00:01    
Ds-Mac-mini:Downloads d.thomaslamourine$ ssh pi@octopi.local
pi@octopi.local's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Dec  9 10:04:08 2016 from 10.0.0.144
pi@octopi:~ $ ~/oprint/bin/pip install OctoPrint-1.3.0.zip

Processing ./OctoPrint-1.3.0.zip
Requirement already satisfied (use --upgrade to upgrade): flask<0.11,>=0.9 in ./oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): werkzeug<0.9,>=0.8.3 in ./oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint==1.3.0)
Collecting tornado==4.0.2 (from OctoPrint==1.3.0)
  Downloading tornado-4.0.2.tar.gz (315kB)
    100% |████████████████████████████████| 317kB 685kB/s 
Collecting sockjs-tornado<1.1,>=1.0.2 (from OctoPrint==1.3.0)
  Downloading sockjs-tornado-1.0.3.tar.gz
Requirement already satisfied (use --upgrade to upgrade): PyYAML<3.11,>=3.10 in ./oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): Flask-Login<0.3,>=0.2.2 in ./oprint/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): Flask-Principal<0.4,>=0.3.5 in ./oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): Flask-Babel<0.10,>=0.9 in ./oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): Flask-Assets<0.11,>=0.10 in ./oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): markdown<2.7,>=2.6.4 in ./oprint/lib/python2.7/site-packages/Markdown-2.6.7-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): pyserial<2.8,>=2.7 in ./oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): netaddr<0.8,>=0.7.17 in ./oprint/lib/python2.7/site-packages/netaddr-0.7.17-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): watchdog<0.9,>=0.8.3 in ./oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): sarge<0.2,>=0.1.4 in ./oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): netifaces<0.11,>=0.10 in ./oprint/lib/python2.7/site-packages/netifaces-0.10.0-py2.7-linux-armv7l.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): pylru<1.1,>=1.0.9 in ./oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): rsa<3.3,>=3.2 in ./oprint/lib/python2.7/site-packages/rsa-3.2-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): pkginfo<1.3,>=1.2.1 in ./oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): requests<2.8,>=2.7 in ./oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): semantic-version<2.5,>=2.4.2 in ./oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): psutil<3.3,>=3.2.1 in ./oprint/lib/python2.7/site-packages/psutil-3.2.1-py2.7-linux-armv7l.egg (from OctoPrint==1.3.0)
Collecting Click<6.3,>=6.2 (from OctoPrint==1.3.0)
  Downloading click-6.2-py2.py3-none-any.whl (70kB)
    100% |████████████████████████████████| 71kB 1.1MB/s 
Requirement already satisfied (use --upgrade to upgrade): awesome-slugify<1.7,>=1.6.5 in ./oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint==1.3.0)
Requirement already satisfied (use --upgrade to upgrade): feedparser<5.3,>=5.2.1 in ./oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint==1.3.0)
Collecting chainmap<1.1,>=1.0.2 (from OctoPrint==1.3.0)
  Downloading chainmap-1.0.2-py2.py3-none-any.whl
Collecting future<0.16,>=0.15 (from OctoPrint==1.3.0)
  Downloading future-0.15.2.tar.gz (1.6MB)
    100% |████████████████████████████████| 1.6MB 150kB/s 
Collecting scandir<1.4,>=1.3 (from OctoPrint==1.3.0)
  Downloading scandir-1.3.zip
[... cut ...]
    Uninstalling OctoPrint-1.2.17:
      Successfully uninstalled OctoPrint-1.2.17
  Running setup.py install for OctoPrint ... done
Successfully installed Click-6.2 OctoPrint-1.3.0 chainmap-1.0.2 future-0.15.2 scandir-1.3 sockjs-tornado-1.0.3 tornado-4.0.2
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
pi@octopi:~ $ 
pi@octopi:~ $ 
pi@octopi:~ $ sudo service octoprint restart
pi@octopi:~ $ 
[image]

foosel, 2016-12-09:
:) You still should debug why you are running into these network issues in the first place though, that sounds like some serious problems with your ISP. Also solving the core issue would allow to update without jumping through these hoops in the future ;)

thomllama, 2016-12-09:
ya, I tried.. was on the phone with Comcast for almost 2 hours.. we did EVERYTHING .. thing is all my other devices don't seem to have an issue.. just the Pi.. I pads, iPhones, Droid phone, Kindles, Macs, 2 (very old, not used anymore but fired up and tested) PC's. all had no issue.. I have a rather fast internet actually (for my location).  I'm thinking it's the actual Pi.. either the comm port, as in the physical plug, or maybe the OS/card needs to be wiped and re-installed.  I'm gunna leave it for now and enjoy just printing.. hoping My brother will have time over the holiday visit to play as I'm sure he'll figure it out in minutes. 👍

thomllama, 2017-01-26:
just so you know.. had to do the same thing again to upgrade to 1.3.1. it just keeps getting booted after taking for ever. Have worked woth Comcast a number of times and nothing else has an issue. websites, cloud based file storage, other FTP stuff,  nothing seems to drag like this does.  not a clue why.   :/