OctoPrint/OctoPrint #1612: Streaming live outside personal LAN

alexanderfrisk, 2016-11-30:
Hi!
I want to start streaming from my webcam to others so they can se what I'm working on.
I'm completely new to programing and octoprint.
Everything works, i can controll my printer through octroprint wirelessly and i also see my printer trough my webcam.
I want to send a link to my friends so they can watch the printer in action aswell. How do I do that?

Sincerely,
Alexander

markwal, 2016-11-30:
https://github.com/foosel/OctoPrint/wiki/FAQ#i-want-to-access-my-octoprint-installation-from-the-internet-how-do-i-do-that

closed 2016-11-30 by markwal

alexanderfrisk, 2016-11-30:
I have read that and it does not make any sense at all for me. I have no clue how to forward port 80 to system or whatever that thing said. Is there any tutorial for a noob any were? Also, I tried ngrok and I got straight into a programing software an as I staden erlier, i don't know programming.
Sincerely, Alexander