OctoPrint/OctoPrint #1613: Unable to hide components in tab view using config.yaml

praveenmax, 2016-12-01:
#### What were you doing?

I am trying to hide certain components such as Terminal, Temperature in the tab view.  As per the docs, config.yaml is the place to do it but when I modified it as below, it still showing those components.

`
--- 
accessControl: 
  salt: x80m4E0IMO0yYB2Wg1q2gTrQWPlywZwz
api: 
  key: FFB44012E080424A82C58E200C47866B
appearance: 
  color: green
  colorTransparent: false
  components: 
    order: 
      generic: []
      navbar: 
        - settings
        - systemmenu
        - login
      settings: 
        - section_printer
        - serial
        - printerprofiles
        - temperatures
        - terminalfilters
        - gcodescripts
        - section_features
        - features
        - webcam
        - accesscontrol
        - api
        - section_octoprint
        - folders
        - appearance
        - logs
      sidebar: 
        - connection
        - state
        - files
      tab: 
        - temperature
        - control
        - gcodeviewer
        - terminal
        - timelapse
      usersettings: 
        - access
        - interface
  defaultLanguage: ~
  disabled: navbar:[], sidebar:[], **tab:[terminal]**, settings:[], usersettings:[], generic:[]
  name: myprinter
feature: 
  temperatureGraph: false
gcodeViewer: 
  enabled: false
plugins: 
  announcements: 
    channels: 
      _important: 
        read_until: 1478614200
  discovery: 
    upnpUuid: 6b364964-dabd-406b-84b4-8de0fdc2816a
  softwareupdate: 
    _config_version: 4
    checks: 
      octoprint: 
        prerelease: false
printerParameters: {}
serial: 
  maxCommunicationTimeouts: {}
  timeout: {}
server: 
  commands: {}
  firstRun: false
  secretKey: 6fV0dxUrRqKWr2VCSNIWpoKp5HAjWf7i
system: {}
temperature: {}
webcam: {}

`

#### What did you expect to happen?

Terminal tab should be hidden.

#### What happened instead?

Terminal tab is still visible

#### Branch & Commit or Version of OctoPrint

Version 1.2.17 (HEAD -> master branch)

#### Browser and Version of Browser, Operating System running Browser

Firefox 

#### Link to octoprint.log

https://gist.github.com/praveenmax/428a61e3eda8db6705123a0a0edd0223

I have read the FAQ.

foosel, 2016-12-01:
That's a know documentation bug ([fixed](b0a6a5675a6132ed8f8cb407317500de6c7ba67f) on the `devel` branch of the docs but not `master`). `disabled` should be under `components`, not next to it.

So like this:

``` yaml
appearance:
  components:
    disabled: ...
```
 instead of this:

``` yaml
appearance:
  components: ...
  disabled: ...
```

praveenmax, 2016-12-01:
@foosel Thanks. Now its working..

closed 2016-12-01 by praveenmax