OctoPrint/OctoPrint #1614: [Request] Machine name at top of window for easy machine identification in multi printer environments.

partsoven, 2016-12-01:
I use Octoprint all day every day on 4 pi's I have running my Lulzbot printers. However keeping them straight on which one is which is always a challenge. 

Currently I configure the top bar to be different colors for each printer. Its nice but it would also be nice to have the machine name/ identifier right at the top of the window next to the Octoprint icon.

If there was a field in the settings to name your printer. You could name your printer and then at the top it would say:
Octoprint- Taz 5

It also might be helpful to have a field in settings for a non default bonjour name, so a user could name the printer "taz-5.local" it would make managing multiple instances of Octoprint so much easier.

AxMod3DPrint, 2016-12-02:
The title box allows you to change the name, when you do this, the tab also shows the title you have given, EG, your printer name. It will show as using your example Taz-5 [Octoprint] in the tab. I have 5 Printers running 5 instances off one Linux Box. It's how I discern between the 5 on my laptop along with pinned tab position on my desktop.

Also, if you're using octopi you can change the name via raspi-config to change the hostname, reboot and you will be able to access it via, again using your example "taz-5.local"

foosel, 2016-12-02:
As already pointed out, that's already possible (and has been for nearly for years I think ;)) through Settings > Appearance.

Changing the bonjour name from OctoPrint is sadly not possible due to technical reasons, you'll need to change the host name for that.

Also OctoPrint is platform agnostic. While it is most commonly ring on a Pi, it is supposed to run in other environment too, so tightly coupling it with something specific as bonjour configuration on raspbian would be a very bad idea.

Closing because already implemented.

closed 2016-12-02 by foosel

partsoven, 2016-12-02:
Thanks just found the Appearance dialog never even knew it was there!