OctoPrint/OctoPrint #1618: [Request] Saving Modified Slicing Settings

cinderblock, 2016-12-03:
I'd like to be able to **save** my tweaks to the Cura slicer settings from within OctoPrint.

Currently, it is easy to edit the settings in a one-off way. They even seem to stick around if the browser window is not reloaded. Unfortunately, there is no "save" button.

#### Current Alternative

1. Open slicer settings in Cura 15.04
1. Edit settings
1. Save settings to ini file
1. Upload ini to OctoPrint

#### Related Issues
 - #447 Seems to have implemented the ability to change the settings in the interface but does not let you save them. (It looks like this fueature was implemented and the issue never closed)
 - https://github.com/foosel/OctoPrint/pull/528#issuecomment-234853130 Looks like others may have been asking for this already.

#### Branch & Commit or Version of OctoPrint

`Version: 1.2.18 (master branch)`

I have read the FAQ. (There was no link to the FAQ in the docs that I could see. I didn't realize the wiki page was for this. Found it with google)

donovan6000, 2016-12-03:
The [M33 Fio](https://github.com/donovan6000/M33-Fio) plugin kinda lets you do this. It lets you save a local copy of a slicer profile with all its current changes, and then that local copy can be uploaded to OctoPrint.
[image]

cp2004, 2022-02-18:
Since Cura was removed from the standard OctoPrint install, this issue doesn't make sense any more. The 'Legacy Cura' plugin is could provide this, but given slicing in OctoPrint is not really worked on anymore I doubt anyone would contribute such a feature.

closed 2022-02-18 by cp2004