OctoPrint/OctoPrint #1619: [Request] REST API to invoke OctoPrint system commands

mcecchi, 2016-12-03:
Hi,

I'm developing an Android App for controlling 3D printer via OctoPrint REST API.

I think would be very useful a REST API for execute SYSTEM commands defined in OctoPrint.

Thanks for your amazing job!

MC

Salandora, 2016-12-03:
Hello @mcecchi

this is already possible.

Here a link to the devel branch documentation
http://docs.octoprint.org/en/devel/api/system.html#execute-a-registered-system-command

This was also possible in master but I can't find it in the documentation so here a link to the python code.
https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/api/__init__.py

Hope it'll help you.

foosel, 2017-01-27:
Duplicate of #804 and as already pointed out also finally documented, now even on master: http://docs.octoprint.org/en/master/api/system.html

closed 2017-01-27 by foosel