OctoPrint/OctoPrint #1620: [brainstorming] Motor disable command changed to a variable

ntoff, 2016-12-04:
I propose a change to the way the "motors off" button works as this has been bugging me for quite a while and with my new found super powers (ahem, I mean, my dodgy copy / paste ability to mess with the source code) I came up with: https://github.com/ntoff/OctoPrint/commits/dev/motorcommand 

It adds a box in the advanced section of the serial connection settings, allowing one to change the default M18 command to anything you like, in the case of Repetier firmware that has to be M84 as M18 does nothing.

I know 1.3 is on a feature freeze but I'm hoping this can be added some time after that as at the moment the default behaviour means us Repetier firmware users need to add a custom control to disable motor hold.

snowdd1, 2016-12-19:
Yeah, means going to terminal, type `M84` <enter>  
Like your idea, very much so.  

What about making it a PR?

ntoff, 2016-12-23:
I kinda gave up on it after reading the reprap wiki, I think it should just be changed to, and hard wired to M84 for all firmwares since Marlin just lists M18 as calling M84 anyway http://reprap.org/wiki/G-code#M18:_Disable_all_stepper_motors and M84 is supported by the majority of firmwares http://reprap.org/wiki/G-code#M84:_Stop_idle_hold

Will wait for @foosel to get back from vacation before pestering her with trivial things like this, I'm sure there's more important things to be fixing.

In the mean time you could install the gcode sidebar plugin which adds a gcode input box to the sidebar to allow you to type in M84 from anywhere no matter which other tab you're on (providing you aren't using touchui or something that rearranges the entire UI)

ntoff, 2016-12-25:
@snowdd1 I've made a little plugin which removes the M18 button and adds an M84 button. May not be the right way or the most elegant way but I've tested it a little bit and it seems to work for me (raspberry pi, octoprint 1.3, chrome on windows) if you want to give it a go and see if it works for you.

https://github.com/ntoff/Octoprint-M18mod

I haven't done extensive testing but it seems to work on vanilla octoprint, and seems to be touch ui compatible.

snowdd1, 2017-01-14:
@ntoff Sorry being a noob with this, how do you install it on my OctoPi?

VBen, 2017-01-14:
@snowdd1 
Just go into the plugin manager an paste the URL https://github.com/ntoff/OctoPrint-M18mod/archive/master.zip there

ntoff, 2017-01-15:
I guess I should have made a more comprehensive install guide, didn't really think anyone but me would ever really use it. Go into the plugin manager where you would normally install plugins and down the bottom where it says "... from URL" paste the url that VBen posted.

snowdd1, 2017-01-16:
Cool, works!

3Dangel, 2017-02-19:
yes it works!! Thanks  a lot!!

foosel, 2017-03-08:
I've been thinking about how to do this (and basically all other commands that are currently hardcoded, see also #154 and #1800), triggered by the recent discussion in #1800. I'm currently leaning towards putting command definitions into printer profiles. Along that line of thought it would also make sense to migrate a lot of the stuff that is now defined globally under Settings > Serial and also Settings > Features and is actually firmware specific and hence better suited for a printer profile. Maybe even the GCODE Scripts (I'm not really sure about that yet though). 

And - not entirely related but somewhat - also allow export and import of printer profiles, to make them way better shareable.

With regards to JUST sending M84 instead of M18, instead of modifying the UI it might be easier to just [replace the command](http://docs.octoprint.org/en/master/plugins/hooks.html#protocol_gcodephase_hook) in the server via a hook. There's even a rewrite example in those docs that could pretty much be adapted by swapping out two values.

@ntoff you've been quite a busy bee with quick fixes both in plugin form and PR form yet I think I still haven't seen a single plugin from you on the repository - why not change that, either with the one you already have or a server side one based on the example mentioned above :)

ntoff, 2017-03-09:
I'm not happy with the hackjobs that are my plugins. I have them on github for anyone else to use if they wish but as I'm not a proficient coder (I literally have NO idea what I'm doing, I'm a metal worker by trade, need me to build you a fence? or panel beat some dents out?), I don't know if I'm doing things in the right way, the most efficient way, and I have no confidence in myself to not screw something up. Pull requests are different, they get looked at by people who know what they're doing any any little changes can easily be implemented.

ntoff, 2017-05-12:
Closing this issue as I've written a plugin that rewrites the motors off command to be compatible with repetier firmware that is now in the official repository: https://github.com/ntoff/Octoprint-M84MotOff

closed 2017-05-12 by ntoff