OctoPrint/OctoPrint #1621: Not able to update in Web UI

Tenderfoot14, 2016-12-04:
#### What were you doing?

I opened octoprint and saw an update was available.

#### What did you expect to happen?

I would update octoprint through the softwareupdate dialog.

#### What happened instead?

Every time (this has happened for multiple versions) there is an update, I am not able to update it from the web ui.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.17 (master branch)

#### Printer model & used firmware incl. version

Printrbot Simple Metal

#### Browser and Version of Browser, Operating System running Browser

Safari, MacOS sierra

#### Link to octoprint.log

http://pastebin.com/hftuSMdF

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

[image]

I have read the FAQ.

gege2b, 2016-12-05:
It's not exactly the same error, but it seems to be related to #1586 (unicode stuff messing with the upgrade)

If it's the case, it should already be fixed in 1.3.0

foosel, 2016-12-05:
The little x next to the update information hints at some misconfiguration being the reason here. Usually that is the case if you haven't configured OctoPrint's checkout folder. If you open up Settings > Software Update, is there a yellow warning message and an exclamation mark next to OctoPrint?

[image]

How does the plugin configuration (click the little wrench icon) look like, especially the configured folder?

[image]

The above is what it should look like under OctoPi for things to properly work.

@gege2b The issue here looks differently - your's was running into trouble during the update itself. This here is a case of the "Update Now" button not even being available due to OctoPrint not having (or thinking to not have) all required information to even perform the update.

Tenderfoot14, 2016-12-06:
There were no errors for lack of a checkout directory, but the directory did not exist. Here is my former plugin config:
[image]

I updated the directory to one on my external drive, and made that directory.
[image]

I restarted the octoprint service, but still no dice. It's also worth noting that I am running Octoprint on a Pi, but not running the OctoPi image.

Kunsi, 2016-12-09:
Checkout directory should be directory where OctoPrint is located. Also should be writeable by OctoPrint-User.

foosel, 2016-12-09:
@Tenderfoot14 How exactly did you install OctoPrint? Are you using OctoPi? If not, what exactly did you do to initially get OctoPrint up and running?

The checkout folder definitely needs to *already exist*, it's not something you can freely configure and then stuff works but the correct value you need to put there depends on how you initially setup OctoPrint. On OctoPi the correct one is `/home/pi/OctoPrint` (and it ships with that preconfigured). If you are running OctoPi and that directory did not exist anymore, something seriously broke in your OctoPi installation.

Finally, this is a support issue, not a bug. Since we already started discussion here let's keep tracking this done here, but in the future with issues like these please use the mailing list or the g+ community because those are the actual support channels. This is a bug tracker reserved for actual bugs in the software.

foosel, 2017-01-27:
Well, apparently we are not handling this here since OP doesn't respond anymore. Closing.

closed 2017-01-27 by foosel