OctoPrint/OctoPrint #1622: [Request] Proxy Camera Stream through OctoPrint

cinderblock, 2016-12-04:
I'm adding a nginx proxy in front of Octoprint in my setup.
The main reason is to be able to monitor the printer remotely but also to clean up the URLs and use https.

Unfortunately, this complicates things with the live streaming view from the camera since it's url is just passed to the browser and the browser opens the connection on its own. This means the configuration needs to be setup for proxied viewing or not, but not both.

This has been covered in a simple way with the suggestion of using HAProxy to combine octoprint and a camera on the same port. I just think this is overkill. It also doesn't work well if you're running multiple instances of OctoPrint on the same machine.

If OctoPrint had an endpoint that was just a proxy for the mjpeg stream available locally, this would make such configuration much easier. This is basically what OctoPrint is doing with the timelapse feature already.

Maybe this functionality would best fit in a plugin?

#### Branch & Commit or Version of OctoPrint

`Version: 1.2.18 (master branch)`

I have read the FAQ.

foosel, 2016-12-05:
> If OctoPrint had an endpoint that was just a proxy for the mjpeg stream available locally

Sadly that would be a very bad idea, simply for performance reasons. You don't want OctoPrint, the same process that is responsible of pushing data as fast as possible to your printer, to be also at the same time trying to push webcam images fetched from an internal network endpoint to connected clients. There's no process prioritization possible within the application threads, so you couldn't even tell OctoPrint to prefer the streaming to the printer over the webcam data processing. And additionally I'm somewhat doubtful that the single threaded asynchronous web server that's utilized here (Tornado) would be up for the task of that plus maintaining several real time websocket connections to connected clients and passing through any incoming requests on the API in a timely manner.

> This has been covered in a simple way with the suggestion of using HAProxy to combine octoprint and a camera on the same port

Not only the same port, but basically the same endpoint, allowing relative URLs to work (which do not work across ports), binding to port 80, and in general making everyone's life easier.

> [haproxy] also doesn't work well if you're running multiple instances of OctoPrint on the same machine.

It works just fine, you just need to tell either haproxy, nginx or whatever else is in front of it to set the correct headers (mainly `X-Script-Name` but also `X-Scheme` when you are also SSL terminating) so that each individual OctoPrint instance will know that it's based not on `scheme://host:port/` but rather on `scheme://host:port/octoprint1/` and configure the webcam stream URL within OctoPrint relatively.

> [reverse proxying the stream] is basically what OctoPrint is doing with the timelapse feature already.

No, it isn't. While there is a proxied endpoint provided for the snapshot, you cannot compare performance requirements for the occasional still image with the requirements for providing a full fledged MJPG stream, maybe even at FullHD and 25fps.

Utilizing a separate process for that (usually in the form of an external webcam server written in C like mjpg streamer) instead of building it into OctoPrint was a conscious design decision due to the restrictions of both the platform, the programming language and also the software architecture.

That being said - I am confused as to why you are not simply using your nginx reverse proxy which you have in place anyhow as substitute for haproxy in your setup. I run such a setup myself (including experiments with nginx' ``X-Accel-Redirect`` header for restricting access to the webcam stream through OctoPrint's access control without actually routing the stream itself through OctoPrint). Could you maybe outline further what exactly you are trying to achieve? Something like:

```
+---------+
|         |
| Client1 +---------------+
|         |               |                                  +--------------------+
+---------+               |                 +---------------->                    |
                          |                 |                |   OctoPrint:5000   |
                        +-v-----------------+-+              |                    |
+---------+             |                     |              +--------------------+
|         +------------->      nginx:443      |
| Client2 |             |                     |              +--------------------+
|         |             +-^-----------------+-+              |                    |
+---------+               |                 |                |    Webcam:8080     |
                          |                 +---------------->                    |
                          |                                  +--------------------+
+---------+               |
|         |               |
| Client3 +---------------+
|         |
+---------+
```

should not be an issue and is in fact even covered in the [example configurations provided in the wiki](https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples), including `X-Script-Name` usage.

Finally:

> Maybe this functionality would best fit in a plugin?

It could be done via a plugin, but all of the above still applies ;)

cinderblock, 2016-12-06:
These are all excellent points. Thank you for the excellent explanation.

I hadn't seriously considered the performance reasons even though I've run into them myself.

The use case I'm looking to support is accessing OctoPrint on the local network directly when possible and through nginx with https on a publicly accessible machine when used remotely. This can be almost transparent to users if the full domain is used and the the local dns is setup properly.

I also wanted to possibly setup multiple instances of OctoPrint on the same machine and didn't want to deal with a crazy HAProxy config.

I ended up just setting up HAProxy on the pi as it does compartmentalize things nicely and the configuration isn't that bad.

foosel, 2016-12-12:
Perfect, closing this then!

closed 2016-12-12 by foosel

KairuByte, 2017-01-17:
Just a side note, I was about to suggest this for security reasons but you bring up good points. Out of curiosity, how does the X-Accel-Redirect method work for access control on the webcam stream? My current solution is VPN only access to the webcam, which is not as convenient as fully public but login needed.

foosel, 2017-01-18:
See [this gist](https://gist.github.com/foosel/81a7449dc1922480bc3e1b67cfa40e5d). In a nutshell, by utilizing a custom tornado handler which when accessed does a check for a user login session. If it's available, it does the X-Accell-Redirect response (which then allows nginx to serve the `internal` marked location), otherwise it just denied access. And OctoPrint itself of course has the tornado route configured for stream and snapshot url instead of the reverse proxy route.

Works fine. Requires work outside of OctoPrint though and a custom plugin. I intend to send a PR to OctoPi though to switch to nginx and this setup, but it probably makes sense to wait with that a bit more until @Salandora and me have had some time to further discuss more granular access permissions - and I also as usual have a lot of other stuff on my plate ;)

KairuByte, 2017-01-24:
Coming from a Windows environment this was a little tricky to get down but it gave me the benefit of easier letsencrypt inclusion. Thank you very much for pointing me in this direction.