OctoPrint/OctoPrint #1623: Loading Model...

bcc20009, 2016-12-04:
I am having a problem with loading any models after inputting the print settings, it simply shows the 'loading model' icon and never loads. I currently have the plugin for the M3D printer. The problem started accruing after i had updated the software.
[image]

donovan6000, 2016-12-04:
This is probably an [M33 Fio](https://github.com/donovan6000/M33-Fio) issue. Make sure your using the latest version of M33 Fio, V1.13, and post any errors that appear in your web browser's JavaScript console when if hangs when loading the model.

foosel, 2016-12-05:
Not an OctoPrint issue but an issue with a third party plugin. Closing for now. Will happily reopen if it proves to be an issue with OctoPrint core after all.

closed 2016-12-05 by foosel

donovan6000, 2016-12-05:
@bcc20009 Feel free to open an issue at [M33 Fio's repo](https://github.com/donovan6000/M33-Fio/issues) so that we can further investigate the issue.

bcc20009, 2016-12-05:
It was a simple issue of the third party not being updated. Though when I checked for updates it stated every thing was fully updated.