OctoPrint/OctoPrint #1625: overwrite userdata on script gcode update

MoonshineSG, 2016-12-05:
#### What were you doing?

Uploading a new version of a pre existing gcode using script. The script adds in user meta data, but the old data is preserved instead of being updated to the new data sent with the new version of the file. 

#### What did you expect to happen?

user meta data to be updated 

#### What happened instead?

old meta data is preserved. 

#### Branch & Commit or Version of OctoPrint

1.3.0rc2 

#### Printer model & used firmware incl. version

n/a

#### Browser and Version of Browser, Operating System running Browser

n/a

#### Link to octoprint.log

n/a

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s) showing the problem:

n/a

I have read the FAQ.

foosel, 2016-12-05:
Can you provide an example request or even better the script?

MoonshineSG, 2016-12-05:
sure. It's on my [repo](https://github.com/MoonshineSG/Simplify3D-to-OctoPrint)

it's lines 93 and 99 of `toctoprint.py`...  basically a curl with "-F" parameter

MoonshineSG, 2016-12-05:
Please ignore this!!!!! It's is entirely my fault! 
I did a manual cleanup of uploaded gcodes (via SSH) and I guess OctoPrint got confused. User issue, not software issue!!! :D

closed 2016-12-05 by MoonshineSG

foosel, 2016-12-05:
Heh, ok!