OctoPrint/OctoPrint #1626: [Request] Show last time for performed prints

bdynamic, 2016-12-05:
For a file which was printed the gcode file gets green and in the extended view there are the following information included:

Name: snowflake_gift_4x.gcode
Uploaded: a day ago
Size: 2.6MB
Filament: 1.50m / 3.61cm³
Estimated print time: 50 minutes
Last printed: a few seconds ago

My feature proposal is, that instead of the "Estimated print time" which can be rather inaccurate (does octopi have a learning module about the print time?) the "Time of last print" should be shown.
This would be helpful for reprinting objects.

#### What were you doing?
Printing a normal print and succeeding in printing

#### What did you expect to happen?
After the print finished the extended view at the file selector doesn't show the last real print time

#### What happened instead?
The time it took for the file to be printed the last time should be shown instead of not so accurate "Estimated print time:"

ntoff, 2016-12-06:
Pretty sure this is already a thing https://github.com/foosel/OctoPrint/pull/1522 Is that what you're suggesting?

foosel, 2016-12-09:
This is from a running OctoPrint 1.3.0 install:

[image]

So yep, already taken care of.

closed 2016-12-09 by foosel