OctoPrint/OctoPrint #1627: playstation eye camera does not respect FPS

JustinHoMi, 2016-12-05:
I'm using a playstation eye camera on a raspberry pi 3. The camera does not respect the framerate settings, and consequently uses 75% of the cpu all the time.

/boot/octopi.txt:

camera_usb_options="-y -q 75 -f 1"

/var/log/messages:

Dec  5 22:11:15 octopi MJPG-streamer [1577]: starting application
Dec  5 22:11:15 octopi MJPG-streamer [1577]: MJPG Streamer Version.: 2.0
Dec  5 22:11:15 octopi MJPG-streamer [1577]: Using V4L2 device.: /dev/video0
Dec  5 22:11:15 octopi MJPG-streamer [1577]: Desired Resolution: 640 x 480
Dec  5 22:11:15 octopi MJPG-streamer [1577]: Frames Per Second.: 1
Dec  5 22:11:15 octopi MJPG-streamer [1577]: Format............: YUYV
Dec  5 22:11:15 octopi MJPG-streamer [1577]: JPEG Quality......: 75
Dec  5 22:11:15 octopi MJPG-streamer [1577]: TV-Norm...........: DEFAULT
Dec  5 22:11:15 octopi MJPG-streamer [1577]: www-folder-path...: ./www/
Dec  5 22:11:15 octopi MJPG-streamer [1577]: HTTP TCP port.....: 8080
Dec  5 22:11:15 octopi MJPG-streamer [1577]: username:password.: disabled
Dec  5 22:11:15 octopi MJPG-streamer [1577]: commands..........: enabled
Dec  5 22:11:15 octopi MJPG-streamer [1577]: starting input plugin input_uvc.so
Dec  5 22:11:15 octopi MJPG-streamer [1577]: starting output plugin: output_http.so (ID: 00)
Dec  5 22:11:32 octopi MJPG-streamer [1577]: serving client: 127.0.0.1

JustinHoMi, 2016-12-05:
#### What were you doing?

Configuring webcam on octopi.

#### What did you expect to happen?

Framerate should be 1fps.

#### What happened instead?

Framerate is 30fps.

#### Branch & Commit or Version of OctoPrint

1.2.18 (also had issue with previous versions)

#### Printer model & used firmware incl. version

Wanhao Duplicator i3 v2.1

#### Browser and Version of Browser, Operating System running Browser

All

#### Link to octoprint.log

http://pastebin.com/BQ15sehY

#### Link to contents of terminal tab or serial.log

not applicable

#### Link to contents of Javascript console in the browser

not applicable

#### Screenshot(s) showing the problem:

not applicable

I have read the FAQ.

foosel, 2016-12-06:
Sorry, but that is not an issue with OctoPrint or even OctoPi, but likely with either mjpg-streamer (not maintained here) or even your camera. There's nothing we can do here :/

foosel, 2016-12-06:
In general though, cameras that require the `-y` parameter to function as mjpg source are a bad choice for an OctoPrint setup running on a Pi or other low-powered CPU, since the webcam server has to transcode the entire camera stream into the proper format first, which of course costs a lot of CPU resources that are then not available for more pressing tasks (such as feeding a printer with data).

You might want to look into getting another camera that supports MJPG natively (my personal favorite here is the Logitech C270), you'll certainly have a better experience with that.

I'm closing this now since it is not a bug within OctoPrint (which this tracker is about) and not even OctoPi and hence we cannot solve this.

closed 2016-12-06 by foosel