OctoPrint/OctoPrint #1628: [Brainstorming] DuetWifi support

Alex9779, 2016-12-07:
I recently upgraded one of my printers with the DuetWifi board.
This board has its own web interface though it just communicates with the firmware with HTTP requests, all GET except of one I think.
The web interface is ok for normal tasks but I am missing some features of Octoprint I got used to like timelapse or print history or Pushbullet notifications and other stuff.

So I was wondering if it would be possible to write a plugin for Octoprint to support this board without using USB connection but using just the available HTTP requests as the builtin web interface does.

I am not yet into Octoprint so deep and just read the tutorial about writing plugins.
This post is intended to get an idea if it would be possible. I don't want to try to implement it and find out in a few weeks that it is not possible...

So my questions are:
- is it possible to overwrite the default connection "module" for USB from a plugin?
- is it possible to change the reading of values for the graphs and print status with a plugin?
- would it also be possible to change the gcode related functions like timelaps on Z change of the gcode visualization so they are not dependent on that Octoprint actually sends the gcode but just uses status messages sent to the firmware to get that information?

I know for some things the firmware has to report extended status message, more than it does now. But this can be solved too.

I just wanna know if someone can imagine that this is possible with the current implementation or if that all is too much change to the core functions and a complete clone of Octoprint would be needed to solve this...

Any comments appreciated...

Alex9779, 2016-12-07:
Just for reference, this is the web interface which is on the DuetWifi: https://github.com/chrishamm/DuetWebControl

foosel, 2016-12-13:
I've been working on and off on a modular redesign of the communication layer for a long long time now (over two years in fact - the "off" times were significantly longer than the "on" times). It's something I definitely want to finally see in 1.4.0 and one of the goals is to get away from the "serial only" point of view. I had a prototype up and printing back in February, see the `dev/commRefactoringNG` branch. Supported serial and TCP, highly extendable regarding both transport layer and protocol talked over it, including flavours. Then stuff happened ;) that somewhat shifted my priorities. I'll need to get that branch merged up with current `devel` again and then see if it still works.

That would probably be the best approach to something like this. Thing is, currently the whole communication layer within OctoPrint is very unforgiving and kinda depends on stuff being a certain way. Which makes the responses to your questions up there basically be "Yes, but it would be a lot of work and quite fragile" or even "No" right now. But I really want to see that changed (and I hope third try will be the charm to finally get this whole comm layer swapped out to enable things like that).

aimartin, 2016-12-14:
It would be a really nice feature... I'm looking at the WIFI module for the MKS board, and using the printer from a Docker image with TCP connection would be really awesome...

Dreaming is free... :P

Regards and thanks for the hard work!

Alex9779, 2016-12-16:
Oh well so I take "No" for the moment. It is ok for me. Not possible means also more time for other things ;)
But I am looking forward for a generalised fronted for my printers, Octoprint works pretty well with the modded firmware for my Ultimaker 2, it did with Marlin but the DuetWifi and RepRapFirmware is a huge step forward...

So how do we handle this? I am fine if you close it down or keep it as a reminder. As more and more printers develop their own web interfaces, I think the UM3 does something similar, and the electronics are developing on to 32 bits it would be really great if you could change the whole communication layer by plugins so you can create/load a plugin for each printer and flavour of communication.
Think this will be a step needed very soon to be able to keep up with the current development of the printers as the serial communication reached its borders...

foosel, 2016-12-16:
I'd actually rather close it, since that revamping of the comm layer is something i don't really need yet another reminder about ;)

As I said, I've been working on that stuff for a long time now (on and off), and the problem is getting it abstracted in such a way that it won't get into the way of things again a couple months down the road. But I really want to have it in 1.4.0. It's a huge endeavor though, and has been a very good approach in the past to find firmware bugs that prevent a whole design from working and have me starting from scratch. And as usual, just having those plugin capabilities in there isn't enough, it also needs devs with targeted printers to develop the plugins - I can't develop things for setups I can't test against :)

Alex9779, 2016-12-16:
Well somehow right though if you supply the basic serial plugin as it is now then I think everything is fine. The community will handle popular platforms itself... Or manufacturers will supply communication plugins for their printers if they just provide a communication interface to their firmware without having to develop a whole web interface...

foosel, 2016-12-16:
I hoped that would be the case when I allowed slicer support to be provided by plugins and look how that turned out ;)

Alex9779, 2016-12-16:
Yeah but the availability of slicers for that platform is limited and honestly the idea is cool though this requires 3D printing to be more evolved as it is now at least in the environment of open source printers. Slicing is an art IMHO and every part requires its own settings to be perfect. This is hard it handle with a integrated slicer and premise static profiles. This will work for simple things and every day printing but not for quality parts...

foosel, 2016-12-16:
I know, but still people demand support for more slicers yet I know of no one actively working on corresponding integrations. My point was rather "demand only won't suffice, I'll need help there once the extendable later is in place or out still will be serial only" :)

Alex9779, 2016-12-16:
Anyways it would be great to have it and I will be up for the challenge to write the plugin for the DuetWifi when the framework is available...

foosel, 2017-03-08:
Forgot to close this back in december. And yes, modular comm layer = still on my mind.

closed 2017-03-08 by foosel

Patrik-svobodik, 2018-11-23:
Hi,  I use DWC but I would like to monitor the print from OctoPrint, would that be possible?