OctoPrint/OctoPrint #1629: RPi 2 - Wifi not working

lkvapil, 2016-12-07:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
that are linked ^-- just up there in the big yellow box. Also read
the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Mark requests with a "[Request]" prefix in the title please. Fully fill
out the bug reporting template for bug reports (if you don't know where
to find some information - it's all described in the contribution
guidelines linked up there in the big yellow box). Do not delete any
lines from the template but those enclosed in [ and ] - and those please
DO delete, they are only provided for your information and removing them
makes your ticket more readable :)

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

lkvapil, 2016-12-07:
Hi,

I have problem with my RPi2 with latest octoprint. I have done all of the common things to successfully run the os, but my problem is, that the RPi2 is not connecting to my WEP network.. I edited the octoprint-network.txt file, filled the data and deleted hashtags in the beginning of lines. I can connect to RPi by ethernet cable, but with the TP-LINK TL-WN722N not. Do I have to install drivers ? 

Thanks for support

Kvapil

markwal, 2016-12-08:
Yes, I believe that adapter requires some significant rigamarole to get it to work with a Raspberry Pi.  I have a TP-Link that I had to find a specific version of the firmware built for my particular version of the kernel.  Googling for the WN722N driver on Raspberry Pi yielded this link: http://elinux.org/RPi_Peripherals#Wireless:_TP-Link_TL-WN722N_USB_wireless_adapter_.28Debian_6.29

I think the kicker there is this bit:

Firmware requirements

If no wlanx device is shown, you might need to download firmware for your USB wifi device (or track down other compatible drivers if they are available). To confirm this, check the dmesg output when you plug in your adapter - typing dmesg at the command prompt may be sufficient - and look for information related to your adapter - the example dmesg output below shows what is seen in the event of the TP-Link firmware not being present - notice that the name of the required firmware file is given (htc_9271.fw), the driver name (ath9k_htc) and an error -22 message:

  usb 1-1.2.4.2: ath9k_htc: Firmware - htc_9271.fw not found
  ath9k_htc: probe of 1-1.2.4.2:1.0 failed with error -22
  usbcore: registered new interface driver ath9k_htc
Armed with this information, try a web search for 'debian 6' and the name of the driver or driver file and/or head over to http://linuxwireless.org. Also check the links at the top of this page.

That wiki is for debian 6 (which is the one before wheezy, OctoPi is now on jessie), but I think it still applies since I don't think the ath9k_htc is part of the image.

This isn't really a "bug" in OctoPrint, so you might want to post on the Raspberry Pi boards or in the Google Community for OctoPrint to see if someone has experience with the same adapter on Jessie.

closed 2016-12-08 by markwal