OctoPrint/OctoPrint #1631: Printing and slicing simultaneously not possible

noahlvb, 2016-12-08:
#### What were you doing?

When a print operation is running, meaning commands are being send to the printer slicing will not give any result. When trying to slice a file while printing operation is running you do net get any feedback and there is no visual response. 

When trying the trigger a slice operation while the an other print operation is running you get the same result, Nothing happens. But you get a 409 error back.

#### What did you expect to happen?

I expected it to continue doing its printing operation and that it started slicing simultaneously.

#### What happened instead?

As described earlier nothing happend.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.18 (master branch)

#### Printer model & used firmware incl. version

Printer: Ultimaker Original plus
Fireware: default Ultimaker fireware

#### Browser and Version of Browser, Operating System running Browser

The issue is not effected by the browser or operating system it is run on.

#### Link to octoprint.log

https://gist.github.com/noahlvb/285792b633a8af53024df7c551765b51

#### Link to contents of terminal tab or serial.log

The Terminal output was the same as when I am just printer and not trying the slice simultaneously.

#### Link to contents of Javascript console in the browser

The JS console did not give an exceptional output.

#### Screenshot(s) showing the problem:

There is literally nothing to see. That is part of the problem.

I have read the FAQ.

foosel, 2016-12-08:
That it doesn't slice while printing is intentional. Slicing (on a Pi) eats a lot of system resources. System resources that are needed to pump data to the printer to actually print. So OctoPrint disables slicing while a print is ongoing (if the slicer plugin in question signals that it slices on the same system, which the bundled CuraEngine plugin does) in order to prevent prints from failing due to resource issues.

Now, OctoPrint should actually disable the slicing controls in its UI for such locally run slicers while printing for that very reason (as should any slicing related plugins for that matter), but apparently that broke sometime ago. I'll take a look.

noahlvb, 2016-12-08:
Thank you for your quick responds. Your explanation sounds really logical.

foosel, 2017-02-24:
@noahlvb Sorry, I somehow missed your follow up question until I opened this ticket to tackle it just now... There is no way to enable slicing on local slicers concurrently without modifying the code of the slicer plugin in question.

----

The problem with the UI inconsistency here has been fixed on the current `maintenance` and soon `devel` branches and will be pushed with the 1.3.2 release. The "Slice" button will now stay disabled if the selected slicer runs locally and a print job is currently running. I also added some more info about that to the slice dialog so this is immediately clear and not a case of "button mysteriously disabled":

[image]

noahlvb, 2017-02-24:
Thank you for your response. I have already resolved my issue by settings up another instance of octoprint on a desktop server to serve as the slicer

foosel, 2017-03-16:
1.3.2 was just released.

closed 2017-03-16 by foosel