OctoPrint/OctoPrint #1633: octoprint 1.3.0 is very picky about commandline order

mrvanes, 2016-12-08:
Thanks for that hard work and making 1.3.0 available!

Took me a while to find out that with the new 1.3.0 this commandline fails:
```/usr/local/bin/octoprint --port 80 --iknowwhatimdoing daemon start```

but this one works:
```/usr/local/bin/octoprint daemon start --port 80 --iknowwhatimdoing```

The logs aren't very vocal about the reason why it failed (it just stops, leaving a pid hanging around) but
```/usr/local/bin/octoprint --port 80 --iknowwhatimdoing serve```

hinted that Octoprint should not be run as root, so I suspected the --iknowwhatimdoing wasn't honoured in that order. Changing the order fixed "serve" and after that "daemon start" as well.

So, 1.3.0 is running nicely, but not without a fuss...

foosel, 2016-12-08:
Ignore the bot, all fine with this issue (special case), will take a look tomorrow.

foosel, 2016-12-09:
Ok, after taking a closer (not sleep deprived ;)) look, this actually looks like it's supposed to. ``--port`` etc are parameters that belong to the sub command ``daemon`` (and ``serve``), so they belong AFTER that command. Look at it as a complete command (even though there's a space in between). For something like ``octoprint-daemon --port`` it would be completely clear why the parameters come after. ``octoprint daemon`` is basically the same idea, just without having to define multiple executables but just use one which supports subcommands.

I initially thought you'd run into problems with the old ``--daemon`` flag to trigger daemon mode instead of the ``daemon`` sub command, and there it would have been a bug, but that appears to work fine (or am I mistaken?).

The new CLI offers more possibilities for future extension, but the price for that is to also be a bit stricter about things to not drown in parameter chaos, so I'm a bit reluctant to make an exception for things like ``octoprint --port 80 daemon`` - it's somewhat backwards ("OctoPrint! Remember port 80, I'll tell you what to do with that info in a sec!") and changing it would set a bad precedent for maintainability in the long run (existing exceptions to the rule right now only are in place to stay backwards compatibility).

nunofgs, 2016-12-10:
I'm on a raspberry pi and having this issue, but I can reproduce it even without arguments:

```
# octoprint
Traceback (most recent call last):
  File "/usr/local/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.0', 'console_scripts', 'octoprint')()
  File "/usr/local/site-packages/octoprint/__init__.py", line 398, in main
    from octoprint.cli import octo
  File "/usr/local/site-packages/octoprint/cli/__init__.py", line 124, in <module>
    from .dev import dev_commands
  File "/usr/local/site-packages/octoprint/cli/dev.py", line 232, in <module>
    @dev_commands.group(name="dev", cls=OctoPrintDevelCommands)
  File "/usr/local/site-packages/click/core.py", line 1169, in decorator
    cmd = group(*args, **kwargs)(f)
  File "/usr/local/site-packages/click/decorators.py", line 115, in decorator
    cmd = _make_command(f, name, attrs, cls)
  File "/usr/local/site-packages/click/decorators.py", line 89, in _make_command
    callback=f, params=params, **attrs)
  File "/usr/local/site-packages/octoprint/cli/dev.py", line 24, in __init__
    from octoprint.util.commandline import CommandlineCaller
  File "/usr/local/site-packages/octoprint/util/__init__.py", line 8
SyntaxError: __future__ statements must appear at beginning of file
```

Any ideas?

foosel, 2016-12-10:
@nunofgs that doesn't look like the same problem. Please open a new ticket. Follow the instructions in the new ticket form please and provide information as requested. Include how you have OctoPrint installed in your system (this looks like a non standard install so please provide information on how you set this up). Also provide a copy of `/usr/local/site-packages/octoprint/util/__init__.py` in that ticket please because the line it appears to be complaining about there is not line 8 in actual OctoPrint 1.3.0 so on first glance something appears fishy with that install.

mrvanes, 2016-12-11:
Ah yes, the reason I bumped into this issue was because my original startup line in rc.local read:
/usr/local/bin/octoprint --port 80 --iknowwhatimdoing --daemon start

After seeing the deprecation warning of --deamon, I just replaced it with:
/usr/local/bin/octoprint --port 80 --iknowwhatimdoing daemon start

Because you know, admins are lazy. And that results in the mentioned problems. I agree that the command -> options paradigm is logical, it just causes issues for lazy people like me.

foosel, 2016-12-12:
@mrvanes all right, going to close this then. Also, I guess you only noticed the deprecation warning in the first place since I managed to break `octoprint --daemon` accidentally (see #1641) :/ and stuff wasn't starting up properly anymore? That has now been fixed too.

closed 2016-12-12 by foosel

foosel, 2016-12-16:
I revisited this after yet another person ran into issues with that.

``octoprint --port 80 --iknowwhatimdoing daemon start`` and the like should now work. I sincerely hope that the decision to change that and going against the default behaviour of the underlying CLI parser won't bite me down the road.