OctoPrint/OctoPrint #1634: REST API to get the whole file list does not list the files inside the folders

anthonyst91, 2016-12-08:
#### What did you expect to happen?

Send the following request in my browser:
http://192.168.1.30/api/files/local?apikey=MY_OWN_API_KEY

Attempt to see all the files in both local & sdcard directories, and also the files present in directories.

#### What happened instead?

The JSON respons lists all the files and the folders in both local and sdcard directories, but unfortunately all the files present in the folders are missing.

I can't see any way to retrieve these files.

#### Branch & Commit or Version of OctoPrint

1.3.0 (master branch)

#### Browser and Version of Browser, Operating System running Browser

All the browsers or third party applications

#### Link to octoprint.log

No logs are needed

#### Link to contents of terminal tab or serial.log

//

#### Link to contents of Javascript console in the browser

//

#### Screenshot(s) showing the problem

Here's a part of the JSON answer :  

{
  "files": [
    {
      "date": 1481238945, 
      "hash": "5a464b3906bb67cc5a864d8efce6d4462d6b6122", 
      "links": [], 
      "name": "cura_catsphere_top.gcode", 
      "notes": [], 
      "origin": "local", 
      "path": "cura_catsphere_top.gcode", 
      "refs": {
        "download": "http://192.168.1.30/downloads/files/local/cura_catsphere_top.gcode", 
        "resource": "http://192.168.1.30/api/files/local/cura_catsphere_top.gcode"
      }, 
      "size": 7301994, 
      "type": "machinecode", 
      "typePath": [
        "machinecode", 
        "gcode"
      ]
    }, 
    {
      "name": "Slic3r", 
      "origin": "local", 
      "path": "Slic3r", 
      "refs": {
        "resource": "http://192.168.1.30/api/files/local/Slic3r"
      }, 
      "type": "folder", 
      "type_path": [
        "folder"
      ]
    }
  ], 
  "free": 1074716672, 
  "total": 7688970240
}

foosel, 2016-12-09:
Documentation bug. You need to set the recursive parameter on that request to true to get all children and children's children.

Eg http://192.168.1.30/api/files/local?apikey=MY_OWN_API_KEY&recursive=true

I'll need to update the docs for that, apparently I managed to forget that even though I did update the data format m)

anthonyst91, 2016-12-09:
Sounds great! This works perfectly with the recursive flag. 
Thanks a lot for the fast reply.

Anthony

foosel, 2016-12-09:
Solved, should soon be available [here](http://docs.octoprint.org/en/devel/api/files.html) (once the doc build runs through) and once 1.3.1 gets released also on the master version of the docs.

Keeping this ticket open until that is the case.

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel