OctoPrint/OctoPrint #1635: [Request] Have octopi.local/#control take viewers to the Control tab type:request

TTN-, 2016-12-09:
Hi,

Right now when go to octopi.local/#control it does not load the control tab, it just stays in the Temperature tab.

When I visit octopi.local and click the control tab, the URL changes to octopi.local/#control 

Would it be possible to have the /#control URL take users straight to the Control tab with the webcam feed?

Is this a bug?

Cheers :-)

Salandora, 2016-12-09:
This is a duplicate of #203

Had something like this done once #461, would need to reimplement it if it is wished.
@foosel would you like to have this feature?

TTN-, 2016-12-09:
Sorry, I did make an effort to prevent duplicate issue filings by searching for #control but didn't fine any similar issues.

I'm pretty keen to have this tbh. Not sure how to implement it though.

foosel, 2016-12-09:
@Salandora now that 1.3.0 is out, fire ahead ;) Do it for 1.3.1/`maintenance` even, qualifies as an improvement I think :)

foosel, 2016-12-09:
Closing this because duplicate, please continue any discussions in #203 :)

closed 2016-12-09 by foosel