OctoPrint/OctoPrint #1637: Unable to delete folder

herrerodani, 2016-12-09:
I recently upgraded to Octoprint 1.3.0 and created a folder named "Imperial Assault". Octoprint renamed it to "Imperial_Assault" and I'm unable to delete it. In octoprint.log it reads:
2016-12-09 12:05:03,101 - tornado.access - WARNING - 409 DELETE /api/files/local/Imperial_Assault (127.0.0.1) 28.17ms

foosel, 2016-12-09:
I said full ticket template ;) But ok. 409 hints at it not being able to delete it because you are currently printing a file from it or similar. Might that be the case?

herrerodani, 2016-12-09:
Thank you for your response. I'm not printing from the folder, but I'm printing a file named "Imperial_Assault_Door_Closedx4.gcode" which is in the root.

#### What were you doing?

I upgraded to 1.3.0 master branch and tried the new folder functionality creating a folder named "Imperial Assault" then tried to delete it. I'm also printing a file in root named "Imperial_Assault_Door_Closedx4.gcode"

#### What did you expect to happen?

The folder to be deleted

#### What happened instead?

Otoprint renamed the folder to "Imperial_Assault" and I'm unable to delete it.

#### Branch & Commit or Version of OctoPrint

1.3.0 master branch

#### Printer model & used firmware incl. version

Prusa i3 Hephestos 1.4.2 BQ Hephestos

#### Browser and Version of Browser, Operating System running Browser

latest Chrome, Ubuntu 14

#### Link to octoprint.log

2016-12-09 12:05:03,101 - tornado.access - WARNING - 409 DELETE /api/files/local/Imperial_Assault (127.0.0.1) 28.17ms

#### Link to contents of terminal tab or serial.log
#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

I have read the FAQ.

foosel, 2016-12-09:
Ah. That is a bug then :D Unrelated to the spaces in the filename though. It thinks because the file that is currently being printed starts with the folder name it is in the folder. Looks like the "is busy" check is a bit too broad there :)

foosel, 2016-12-09:
And fixed by above commit on `maintenance`, will be released with 1.3.1(rc1). Marking as solved and keeping this open until this is the case

herrerodani, 2016-12-09:
Yeah, that was it. Once the Printer ended printing "Imperial_Assault_Door_Closedx4.gcode" I could delete "Imperial_Assault" folder. Thank you very much for your work.

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel