OctoPrint/OctoPrint #1638: [Request] -Optional- confirmation dialogue before cancelling a print

rob-miller, 2016-12-10:
Just released OctoPrint 1.3.0 added a confirmation dialogue before cancelling a print.  Please make this optional?  I have reviewed the original request #1557 and understand the intent, however I do frequently cancel prints on purpose because of problems and the added dialogue is frustrating, and have not (yet) had an example where I accidentally cancelled a print in this way.  

Frustration is definitely a lower priority that prints ruined for no good reason, but I suspect the frustration is much more frequent now?  Thank you.

ntoff, 2016-12-10:
This seems to do the trick https://github.com/ntoff/OctoPrint/commit/e458e3dd184e444e4b7b685c109b9c38c9d71208 Tested it here and it seems to work, not sure if it's pull request quality though. @foosel?

HWiese1980, 2016-12-11:
I have cancelled prints accidentally and I think the confirmation dialog is a very good idea. On the other hand I do understand that it can be annoying when you frequently cancel prints. One way to do it could be to suppress the dialog when you press SHIFT or CTRL or something while clicking on the button. Just a suggestion. I think one would not so easily press a key on the keyboard and click the cancel button all together by accident.

ntoff, 2016-12-11:
What about those of us on tablets or phones that don't have a shift key?

Cyberman-tM, 2016-12-11:
Is there a way to disable the confirmation via a config file perhaps? I too need the "cancel" button because something is wrong and I don't want to risk damage to the printer.

rob-miller, 2016-12-11:
My thinking was more along the lines that the core use case is to cancel the print immediately because something is actively going wrong (accepting that there is a queue of commands still to process).  I may not have both hands free, or may be monitoring the print on a phone.  As a personal preference I find the confirmation dialogue frustrating, and I can clear my screensaver without a blind mouse click (my reading of the original issue in #1557).  I'm very happy if the default is to have the confirmation dialogue, but for me I prefer to cancel directly with an unmodified button press.

ntoff, 2016-12-11:
>and I can clear my screensaver without a blind mouse click

I think it's more for people on touch screens that blank the screen but leave the touch part active to wake it up like the raspberry pi does with its built in touch screen. I'm not sure if touching the screen actually does anything other than waking it up but it's not like a phone or tablet where you have an unlock button, you just touch the screen to wake it.

nophead, 2016-12-11:
Those of us that use PCs with disabled screensavers do not want an extra dialog. It is much better to cancel immediately. If I ever hit the button by mistake that is my fault not Octoprints.

foosel, 2016-12-12:
@ntoff looks good, please send a PR!

ntoff, 2016-12-12:
ok, pull request sent. One thing I completely forgot about is the translation file for that line, is that something I need to do? Or is that for the work of whoever translates it? I don't speak anything but English (and bad English) but I don't have to create placeholders or anything in the translation file do I?

foosel, 2016-12-13:
And merged, even against `maintenance`. So that option will be available in 1.3.1. I moved the checkbox in the settings up a bit towards the other more UI related ones.

Regarding the translation stuff, you have done everything correctly. On every release I extract all translatable strings via `python setup.py babel_refresh`, update the german translation, compile that (`python setup.py babel_compile`) and bundle it so it gets included in the actual installations (`python setup.py babel_bundle -lde`). And german is the only language bundled with OctoPrint at the moment, simply because I only speak english and german well enough to trust myself to provide proper translations ;)

ntoff, 2016-12-13:
>I moved the checkbox in the settings up a bit towards the other more UI related ones.

I actually contemplated doing a last minute change on it and moving it up, renaming it to "Enable print cancel confirmation" and putting it up top with all the other "Enable X" things but sort of figured "confirm before cancelling" sounded a bit more straight forward so I left it.

Good to know about the translation thing, I did wonder how it knew which lines contained which texts and whether someone was sitting there manually adjusting things every time someone added a new feature.

snowdd1, 2016-12-19:
Though this issue is already closed - I was about to create a request and stumbled upon this one. I found it definitely confusing to need to "proceed" in a cancel confirmation box.  
When first time needing to cancle a job I accidently canceled the cancel instead of the print.  
IMHO the dialog buttons should be labeled "Cancel Print" and "Continue Print".  

[These are my own re-translations, as I am using octoprint in German]

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel