OctoPrint/OctoPrint #1640: [Request] / Idea - Mouse over the temperature graph for an accurate display of the temperature at that point

ntoff, 2016-12-10:
Much like how this works http://www.flotcharts.org/flot/examples/tracking/index.html Is it possible for this kind of functionality to be added to the temperature graph or does it work in a way that makes this impossible? Just looking at my temperature graph and there are very slight fluctuations in it, would be handy to hover over points to see just how far it's fluctuating.

foosel, 2016-12-13:
@ntoff got curious, did a first throw at it in the above commit/branch. Hovering over the graph enables the crosshair. Temps at that point will be shown in italics in the legend, replacing the current values. Moving the mouse away from the graph again resets the displayed legend values to the current ones.

[image]

Seems to be causing problems on mobile though. I'll have to investigate that next year ;)

ntoff, 2016-12-13:
Nice. I guess it would have to be disabled for mobile?

Might also be a while before I do anything more, every time I open "my computer" on my spare PC (which is where the magic happens) it takes ages for C drive to show up so I think the hard drive might be dying, it's a really old hard drive, and it's doing weird *weird* things. I'm trying to modify the repetier eeprom plugin and it's loading the python files from c:/oprintdev but it's loading the jinja and javascript files from the github directory in my documents o_0. 

Thanks for all the hard work making Octoprint so great, have a safe Christmas.

ntoff, 2017-06-06:
Can I close this, since there's a plugin that exists now? http://plugins.octoprint.org/plugins/tempsgraph/

foosel, 2017-06-07:
Leave it open for a bit longer. I still have that commit I did in December, and a quick look just now showed me why stuff was broken on mobile with that. Now the only thing left is to figure out why things look weird under TouchUI ;)

foosel, 2017-10-17:
1.3.5 was released yesterday.

closed 2017-10-17 by foosel