OctoPrint/OctoPrint #1644: Wrong visualization in GCode viewer

sanchosk, 2016-12-11:
#### What were you doing?

Printing from Slic3r direct upload

#### What did you expect to happen?

When looking at the visualization, I expected the layers to go one by one as they should.
First layer should be 0.4mm, the rest is 0.3mm in size, so 0.7, 1, 1.3, ...
Here is the source [G-code file](http://sancho.sk/problem.gcode).

#### What happened instead?

The order of layers is mixed.
Layer 1: 0.4mm (OK)
Layer 2: 1mm (bad)
Layer 3: 1.3mm (bad)
Layer 4: 1.9mm (bad)
Layer 5: 2.5mm (bad)
Layer 6: 3.1mm (bad)
Layer 7: 1.6mm (bad)
...
As you can see, it jumps up and down.
When using Gcode analyzer (gcode.ws), everything is OK. Manual verification of the g-code source seems also OK.
The print was cancelled in panic once, but after manual verification of G-code I started it again and it seems to print OK.

#### Branch & Commit or Version of OctoPrint

Version: 1.3.0 (rc/devel branch)

#### Printer model & used firmware incl. version

RepRap i3 clone from Hestine (M505 or something), Repetier firmware 0.92.1

#### Browser and Version of Browser, Operating System running Browser

Firefox 50.0.2, Ubuntu 14.10

#### Link to octoprint.log

Important content in serial.log below.

#### Link to contents of terminal tab or serial.log

I have the log from the last 1-2 days, it's a bit big. The file is still printing, so read from the bottom of the file. It can be found [here](http://sancho.sk/serial.log).

#### Link to contents of Javascript console in the browser

https://gist.github.com/sanchosk/ceec21d7115eee35ab725343f6497d64

#### Screenshot(s) showing the problem:

Layer 6:
[image]

Layer 7:
[image]

I have read the FAQ.
Yes, I did :)

sanchosk, 2016-12-11:
Seems like a problem with caching here.
When I cleared cache and everything, the page loaded and displayed OK.
Then I wanted to report here, so I got back to github, opened new tab for OctoPrint and bum - again bad order. Even at layer 17 I clicked on "Next Layer" and got to the very last one (61 or something).
I love these non-reproducible errors :(

foosel, 2016-12-12:
Just for the record, cannot reproduce this here either right now (using the provided file).

In general, what the gcode viewer is displaying is what it parsed from the file it downloaded from the server (which should be exactly what you uploaded). It downloads the file, parses it, generates an internal representation of the model and then renders that. That representation boils basically down to one huge array of layers and for each layer a list of contained moves at that height (with and without extrusion).

I'm not really sure how it would be possible for that data to become jumbled and hence the layer height becoming wrong to be honest. When that issue arose, did the moves of the layer itself look correct in the viewer and it was only the reported height that was off, or were those moves also from a wrong layer?

sanchosk, 2016-12-12:
Yes, the moves were OK per layer, just the order of layers was messed up. The layer Z height was also OK, the only problem was the order. Also, as you can see on the screenshots, the final layer rendering is OK for both the layer 6 and 7. They are just not where they should be from the Z perspective :(

sanchosk, 2016-12-28:
As it did not reproduce nest week and does not reproduce now with 1.4rc, I am closing this.

closed 2016-12-28 by sanchosk