OctoPrint/OctoPrint #1645: Feature request: Support for Repetier JAM report

sanchosk, 2016-12-11:
This is NOT an issue.
I am reading through Repetier firmware source.
I found out that there is a "jam detection" facility.
In case of jam, the firmware does multiple things:
- if there is Display, it will start UI dialog to clear jam
- if it is printing from SD, it will pause the print
- it will report back to host status "RequestPause:Extruder Jam Detected!"

It would be fantastic, if OctoPrint can support the last mentioned - when the reply is as mentioned, it would pause the print.
Then, there is even a jam-clear G-Code (M513) that can be sent back to printer as a resume action.

My plan is to "hack" the repetier firmware a bit to allow jam functionality to be used as a basic filament sensor response - when a microswitch disconnects circuit when filament runs out, the firmware will initiate the "jam" action and report back to the host.
This will allow me to save a lot of multi-hour prints.
Is it difficult to implement this? Is there already different filament sensor support in OctoPrint?

foosel, 2016-12-12:
Together with the Smoothie guys we actually implemented a specific type of message for these kind of things, so called action commands. See also [OctoPrint's docs](http://docs.octoprint.org/en/master/features/action_commands.html). That's exactly for things like that. 

It would be possible to create a plugin for also reacting to this firmware specific message, but I'd rather not start implementing every competing standard into the already way too complex communication layer :/

sanchosk, 2016-12-12:
Other solution would be to mod the firmware (easy) to pass correct messages back (as I already found the part in the code, that should be relatively easy).
But the plugin idea seems more plausible - extensible for other firmwares, too.
Looking forward to your plugin-how-to video :D

closed 2016-12-12 by sanchosk

foosel, 2016-12-12:
@sanchosk definitely going to happen - probably not in 2016 though, the last couple of months are starting to take their toll for good I fear :)

sanchosk, 2016-12-12:
Seriously - try to use the advantages of being self-governed thanks to sources like Patreon etc.
Relax a bit.
I think majority of supporters will agree with me - I am not supporting you to work 24/7 (weekend live-streams, etc.). I want you to enjoy your life AND still be able to work on OctoPrint.
Without the hassle of going to office every morning, wake up early or only have 20 days of vacation per year. I expected decrease of activities in December (most of financial institutions suspend their IT operations in this time, so I am used to this rhythm myself). Instead, you came with 2 releases and lots of bugfixes. While I appreciate all of it, I would suggest to slow down the pace to enjoy it a bit more :)
Happy holidays :)

foosel, 2016-12-12:
@sanchosk I think there's something in my eye. Thank you for these words and happy holidays to you too! :)

sanchosk, 2016-12-28:
Well, the job is done - I modified Repetier messaging to send the original message back (info and error) and added a new line with 
// action:pause
And it WORKS :)