OctoPrint/OctoPrint #1646: [REQUEST] Convert TAB words to ICONS instead? (Temperature, Control, GCode Viewer...)

bolsoncerrado, 2016-12-11:
So heading to XXIst century OctoPrint, how about turning the TAB names into ICONs (user selectable??)?

The thing is, if you install more than 2 plugins you end up having to do extra clicks to access sometimes most useful plugins hidden in a list.

Turning them icons would provide much faster access to much more options with a single click.

LazeMSS, 2021-03-31:
Can be done by plugins:
https://plugins.octoprint.org/plugins/uicustomizer/
https://plugins.octoprint.org/plugins/themeify/

closed 2021-03-31 by LazeMSS

jneilliii, 2021-03-31:
I'd like to point out that Tab Order also implements icons for tabs and you can hide the name.

https://plugins.octoprint.org/plugins/taborder/