OctoPrint/OctoPrint #1647: Error parsing command-line arguments

nunofgs, 2016-12-11:
#### What were you doing?

Attempting to run version 1.3.0 in a Docker container. Here's an example _Dockerfile_:

```Dockerfile
FROM python:2.7-slim

RUN apt-get update \
  && apt-get install -y build-essential git \
  && git clone -b 1.3.0 https://github.com/foosel/OctoPrint.git \
  && cd OctoPrint \
  && python setup.py install

CMD ["octoprint"]
```

#### What did you expect to happen?

When running `octoprint --help` I expect to see the _usage_ message.

#### What happened instead?

```
# octoprint --help
Usage: octoprint [OPTIONS] COMMAND [ARGS]...

Error: Invalid value for "--version": 1.3.0 is not a valid boolean
```

Running without arguments also yields the same result.

#### Branch & Commit or Version of OctoPrint

1.3.0

foosel, 2016-12-12:
Testing the same command on OctoPi:

```
(oprint)pi@octopi:~/OctoPrint $ octoprint --help
Usage: octoprint [OPTIONS] COMMAND [ARGS]...

Options:
  -b, --basedir PATH  Specify the basedir to use for uploads, timelapses etc.
  -c, --config PATH   Specify the config file to use.
  -v, --verbose       Increase logging verbosity
  --safe              Enable safe mode; disables all third party plugins
  --version           Show the version and exit.
  --help              Show this message and exit.

Commands:
  client   Basic API client.
  config   Basic config manipulation.
  daemon   Starts, stops or restarts in daemon mode.
  dev      Additional commands for development tasks.
  plugins  Additional commands provided by plugins.
  serve    Starts the OctoPrint server.
```

Same behaviour on any other checkout, regardless of OS.

Also tried a replication of your exact checkout & install steps. Differences being: no docker, virtual environment:

```
$ git clone -b 1.3.0 https://github.com/foosel/OctoPrint.git OctoPrint-1.3.0-Test && cd OctoPrint-1.3.0-Test && virtualenv venv && source ./venv/bin/activate && python setup.py install
[...]
$ octoprint --help
Usage: octoprint [OPTIONS] COMMAND [ARGS]...

Options:
  -b, --basedir PATH  Specify the basedir to use for uploads, timelapses etc.
  -c, --config PATH   Specify the config file to use.
  -v, --verbose       Increase logging verbosity
  --safe              Enable safe mode; disables all third party plugins
  --version           Show the version and exit.
  --help              Show this message and exit.

Commands:
  client   Basic API client.
  config   Basic config manipulation.
  daemon   Starts, stops or restarts in daemon mode.
  dev      Additional commands for development tasks.
  plugins  Additional commands provided by plugins.
  serve    Starts the OctoPrint server.
```

The error sounds like something seriously messed up the command line arguments here.

Now, I have to admit that I'm completely unfamiliar with Docker, and especially how it works internally, but could it be that it's doing something in its environmental setup or they way the actual executable gets called that's messing with things here?

I played around a bit with things and finally could reproduce the exact error message:

```
$ OCTOPRINT_VERSION=1.3.0 octoprint --help
Usage: octoprint [OPTIONS] COMMAND [ARGS]...

Error: Invalid value for "--version": 1.3.0 is not a valid boolean
```

Could it be that such an environment variable is set by Docker, causing the issues here? Should be easy enough to check with ``CMD ["env"]`` if I understand things right. If so, that would be the reason. Click (which is what I use for argument parsing) here is set up to also take arguments from the environment, prefixed with ``OCTOPRINT_``. Apparently it also does that for the flag-parameter version (which I'll have to see if I can prevent that from happening), causing this issue. The question though would be via such an environment variable would even be set in the container in the first place.

foosel, 2016-12-12:
For the record, with above commit I cannot reproduce anymore:

```
$ OCTOPRINT_VERSION=1.3.0 octoprint --help
Usage: octoprint [OPTIONS] COMMAND [ARGS]...

Options:
  -b, --basedir PATH  Specify the basedir to use for uploads, timelapses etc.
  -c, --config PATH   Specify the config file to use.
  -v, --verbose       Increase logging verbosity
  --safe              Enable safe mode; disables all third party plugins
  --version           Show the version and exit.
  --help              Show this message and exit.

Commands:
  client   Basic API client.
  config   Basic config manipulation.
  daemon   Starts, stops or restarts in daemon mode.
  dev      Additional commands for development tasks.
  plugins  Additional commands provided by plugins.
  serve    Starts the OctoPrint server.
```

That is available on the `maintenance` branch and will be released with 1.3.0.

Would still be important to know if that was actually the reason behind this in the Docker environment and why that var was even set.

foosel, 2016-12-14:
@nunofgs could you report back on if the above fix works for you?

nunofgs, 2016-12-15:
@foosel: confirmed. Works perfectly for me!

foosel, 2016-12-15:
Perfect, marking this as solved then, will be released with 1.3.1 as already mentioned.

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel