OctoPrint/OctoPrint #1648: Add websocket-client as a dependancy

chickenchuck040, 2016-12-12:
#### What were you doing?

Trying to use the `octoprint client listen` command. 

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

I expected the `octoprint client listen` to show messages from the Octoprint server

#### What happened instead?

It gave me an error message about `no module named websocket`
```
➜  octoprint git:(master) octoprint client listen               
Traceback (most recent call last):
  File "/usr/bin/octoprint", line 9, in <module>
    load_entry_point('OctoPrint', 'console_scripts', 'octoprint')()
  File "/usr/lib/python2.7/site-packages/OctoPrint-1.3.0.dev1245+gcba742a-py2.7.egg/octoprint/__init__.py", line 310, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/usr/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/OctoPrint-1.3.0.dev1245+gcba742a-py2.7.egg/octoprint/cli/client.py", line 176, in listen
    on_message=on_message)
  File "/usr/lib/python2.7/site-packages/OctoPrint-1.3.0.dev1245+gcba742a-py2.7.egg/octoprint_client/__init__.py", line 382, in connect_socket
    socket.connect()
  File "/usr/lib/python2.7/site-packages/OctoPrint-1.3.0.dev1245+gcba742a-py2.7.egg/octoprint_client/__init__.py", line 124, in connect
    self._prepare()
  File "/usr/lib/python2.7/site-packages/OctoPrint-1.3.0.dev1245+gcba742a-py2.7.egg/octoprint_client/__init__.py", line 59, in _prepare
    import websocket
ImportError: No module named websocket
```

I found that after installing the `websocket-client` module, it works as expected.
I also found no reference to it in `setup.py`

#### Branch & Commit or Version of OctoPrint
Version 1.3.0 from the master branch

#### Printer model & used firmware incl. version
Marlin
Not applicable

#### Browser and Version of Browser, Operating System running Browser
Not applicable

#### Link to octoprint.log
https://gist.github.com/chickenchuck040/cba0f5c30e0d95481d8111a4d145d1c9

#### Link to contents of terminal tab or serial.log
Not applicable

#### Link to contents of Javascript console in the browser
Not applicable

#### Screenshot(s) showing the problem:
Not Applicable

I have read the FAQ.

foosel, 2016-12-13:
Ouch. Sorry for that and thanks for the heads-up

foosel, 2016-12-13:
Solved in above commit. Will be in 1.3.1

chickenchuck040, 2016-12-13:
Thanks for the quick response!

foosel, 2017-01-25:
Part of 1.3.1 which has just been released.

closed 2017-01-25 by foosel