OctoPrint/OctoPrint #1651: Unable to update from 1.2.16

jonaslorander, 2016-12-13:
#### What were you doing?

Trying to update OctoPrint 1.2.16 to 1.3.0 by clicking the update button in the announcement.
(Have tried all version from 1.2.18 when they arrived)

#### What did you expect to happen?
A successfull update of OctoPrint.

#### What happened instead?
"Update failed!" notificaion appeared.

#### Branch & Commit or Version of OctoPrint
Version: 1.2.16 (master branch)

#### Printer model & used firmware incl. version
N/A

#### Browser and Version of Browser, Operating System running Browser
Google Chrome 54.0.2840.99 m on Windows 7

#### Link to octoprint.log
https://gist.github.com/jonaslorander/16cc238561b39842d2366ca82f18f744

#### Link to contents of terminal tab or serial.log
https://gist.github.com/jonaslorander/80b58a0cd313b97d34214ca4b6f0f746

#### Link to contents of Javascript console in the browser
https://gist.github.com/jonaslorander/c6e77f895f2514efd682dc15efaf78dd

#### Screenshot(s) showing the problem:

I have read the FAQ.

According to FAQ it should be as simple as saving the settings. But it does not work. You can see at 2016-12-13 09:48:38,196 in the octoprint.log that the saved occured.
https://github.com/foosel/OctoPrint/wiki/FAQ#im-running-1216-i-tried-to-update-to-a-newer-version-via-the-software-update-plugin-but-i-get-an-error

foosel, 2016-12-13:
Your issue is not the bug described in the FAQ, but rather a wrong configuration from the looks of it:

```
2016-12-13 09:43:39,473 - octoprint.plugins.softwareupdate.updaters.update_script - WARNING - Target: octoprint, update stderr:
Not a git repository
```

Whatever you have configured as OctoPrint's checkout folder isn't actually OctoPrint's checkout folder with the `.git` subfolder and everything. Judging by the rest of the log it should *probably* be set to `/home/pi/OctoPrint`, basically like this:

[image]

Is it and is `/home/pi/OctoPrint` actually still a valid git repository?

jonaslorander, 2016-12-13:
That did the trick! I had `/home/pi/tmp` for some reason. 

Thanks!

closed 2016-12-13 by jonaslorander