OctoPrint/OctoPrint #1652: [Request] Allow txt and bin files to be uploaded to SD card

macklij, 2016-12-13:
FEATURE REQUEST :)  Thanks

#### What were you doing?
Uploading files to SD card on a Smoothieboard. 
It would be really useful to be able to upload Smoothie's 'config.txt' and possibly 'firmware.bin' files as well as gcode files.  It would mean not having to unplug RPi from USB, take laptop etc etc. Especially if printer not near PC.

#### What did you expect to happen?
N/A This is a feature request

#### What happened instead?
N/ A This is a feature request

#### Branch & Commit or Version of OctoPrint
Version: 1.3.0 (master branch)

#### Printer model & used firmware incl. version

Homebrew using Smoothieware edge-67791da

#### Browser and Version of Browser, Operating System running Browser

Chrome 54.0.2840.99 m

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N?a

I have read the FAQ.

foosel, 2016-12-14:
Prime example of something that could be done by a plugin :) The list of allowed extensions can be enhanced through [this hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-extension-tree), and a bit further digging would also allow something like a comfortable upload/editor interface for the relevant files. The actual file transfer would need to circumvent one final check within OctoPrint which currently can't be overwritten by a plugin, but that would be a matter of just reimplementing an existing method on the printer interface without that check.

The problem with the available SD handling routines in firmware in general (and afaik also in Smoothieware) is that while there's a command to transfer a file *to* the printer, there isn't one to receive an existing file's contents *from* the printer. So this would boil down to something like a "write only" interface, which is a bit sad.

I have to admit that while I have a Smoothieboard, it's currently not built into any of my printers, so I wouldn't be able to test something like this. Plus I have my hands full with the regular OctoPrint related tasks. I'd be happy to lend a helping hand though to anyone who wants to implement something like this in form of a SmoothiewareConfig plugin or something along these lines!

ntoff, 2016-12-23:
What about adding .hex to that list for arduino boards (for whoever takes this on). I know there's a firmware updater plugin but it doesn't give much feedback about what's currently going on so I'd rather use avrdude via ssh. My current solution is a samba share that I can upload firmware to then ssh in and run avrdude on it

jneilliii, 2017-01-20:
The latest version of [flashArduino](https://github.com/Leapfrog3DPrinters/OctoPrint-flashArduino) displays pretty detailed avrdude ouptut duting the process.

jneilliii, 2017-09-17:
I took a stab at this plugin and it is working for uploading txt and bin files locally, but not sure if it's working for transfer to SD.  You can test it by installing from the repository at the link below. I only tested with a couple of small files, so there may be issues with larger firmware files, but should be equivalent to transferring an stl file as I used the model dictionary type.

https://github.com/jneilliii/OctoPrint-SmoothiewareConfig

macklij, 2017-09-18:
Hi there

Thank you so much for your time and effort with this. I am away from home until Friday but will certainly play this weekend and let you know. 

Best

Justin

iPhone  iType  iApologise

> On 18 Sep 2017, at 00:04, jneilliii <notifications@github.com> wrote:
> 
> SD. You can test it by installing from the repository

jneilliii, 2017-11-04:
@foosel I'm experiencing an odd behavior with the plugin I created for this.  It works great for uploading txt to local storage, but when uploading to SD it just freezes. I'm just using the get_extension_tree hook. Any suggestions?

foosel, 2017-11-06:
@jneilliii I replied in your own repo.

LazeMSS, 2021-03-31:
Possible by plugin:
https://plugins.octoprint.org/plugins/uploadanything/

closed 2021-03-31 by LazeMSS